openshwprojects/BK7231GUIFlashTool #2: Rewrite previously read original firmware

ipn00tic, 2023-01-05:
Great program.
It was very interesting and I think it should be allowed to go back to the original firmware via dump.
The program makes the original dump but has no option to write the original dump in case something goes wrong.

thanks for the work

openshwprojects, 2023-01-05:
Thanks, it's a good idea and it will be added soon, I think.

openshwprojects, 2023-04-19:
Added, thanks!

closed 2023-04-19 by openshwprojects

BenJamesAndo, 2024-11-18:
In case anyone is looking for this option, you have to click `Show advanced options`, then tick `Allow backup restore`. After that you should be able to select your Tuya backup .bin file in `Select firmware:`
The flasher tool looks in the backups folder. I was able to succesfully re-flash a Tuya firmware file.

[image]