openshwprojects/BK7231GUIFlashTool #4: "Do firmware write (no backup!)" does nothing

Suxsem, 2023-01-06:
All buttons work except for "Do firmware write (no backup!)", can you please take a look?

NielsPiersma, 2023-01-06:
I have the same issue. I can't "forcibly" overwrite my test boards. I tested for Both T and N boards.
Niels

NielsPiersma, 2023-01-06:
First attempt 
Backup goes fine, flashing fails.
[image]

NielsPiersma, 2023-01-06:
second attempt, with high baud-rate
[image]

NielsPiersma, 2023-01-06:
and with higest baud rate
[image]
Will revert back too lower baud rate and test further

NielsPiersma, 2023-01-06:
Test read/write (advanced option) works fine.
[image]

Did notice log screen doesn't resize.

NielsPiersma, 2023-01-06:
And the last attempt for the N was successful. 
[image]

Now will try the T version.
Niels

openshwprojects, 2023-01-06:
I simply forget to link action to the button, will push a fix and release now. Thanks.

openshwprojects, 2023-01-06:
https://github.com/openshwprojects/BK7231GUIFlashTool/commit/4e605dc3f314a29a9fc8573524e6b54ade5c54fb

closed 2023-01-06 by openshwprojects