openshwprojects/BK7231GUIFlashTool #5: [BK7231N][CB3S] Reading 0x00... failed with serial.BytesToRead 4056

Rafik89, 2023-01-06:
Hello, 

BK7231N (CB3S) chip I can't do flasch or backup. Whichever I do. 

In the HID Download itself goes without the slightest problem. 

```
Starting read!
Now is: piątek, 6 stycznia 2023 12:18:40.
Flasher mode: BK7231N
Going to open port: COM4.
Serial port open!
Getting bus... (now, please do reboot by CEN or by power off/on)
Getting bus failed, will try again - 0/100!
Getting bus failed, will try again - 1/100!
Getting bus failed, will try again - 2/100!
Getting bus failed, will try again - 3/100!
Getting bus failed, will try again - 4/100!
Getting bus success!
Going to set baud rate setting (921600)!
Will try to read device flash MID (for unprotect N):
Flash MID loaded: 15701C
Will now search for Flash def in out database...
Flash def found! For: 15701C
Flash information: mid: 15701C, icName: EN25QH16B, manufacturer: ESMT, szMem: 1000000, szSR: 1, cwUnp: 0, cwEnp: 7, cwMsk: 3C, sb: 2, lb: 4, cwdRd: 05-FF-FF-FF, cwdWr: 01-FF-FF-FF
Entering SetProtectState(True)...
sr: 5c
final sr: 5c
msk: 3c
cw: 0, sb: 2, lb: 4
bfd: 0
sr: 40
final sr: 40
msk: 3c
cw: 0, sb: 2, lb: 4
bfd: 0
SetProtectState(True) success!
Going to start reading at offset 0x00...
Reading 0x00... failed with serial.BytesToRead 4056
Failed!
```

```
>python uartprogram firmware.bin -d com4 -r
UartDownloader....
Read Getting Bus...
Gotten Bus...
Set baudrate successful
len: 119000
startAddr: 11000
Reading 11000
....
ReadSector Success 118000 len 1000
1150976
CRC should be b7d1bc8d
CRC is 6ce0b681
CRC check failed
Wrote 119000 bytes to firmware.bin

```

openshwprojects, 2023-01-06:
Thanks, will you be able to recheck if I give you updated flasher binary?

Rafik89, 2023-01-06:
Yes, i can check this. First afternoon 😉

az-src, 2023-01-06:
The wb3s device also has a reading problem. The recording goes smoothly.

Starting read-write test!
Now is: 6 января 2023 г. 23:04:00.
Flasher mode: BK7231T
Going to open port: COM6.
Serial port open!
Getting bus... (now, please do reboot by CEN or by power off/on)
Getting bus failed, will try again - 0/100!
Getting bus failed, will try again - 1/100!
Getting bus success!
Going to set baud rate setting (115200)!
Going to do erase, start 69632, sec count 2!
Erasing sector 69632... ok! Erasing sector 73728... ok! 
All selected sectors erased!
Going to start reading at offset 0x11000...
Reading 0x11000... failed with serial.BytesToRead 4050
Failed! Read failed?

itsj4y, 2023-01-06:
Also having a problem flashing a WB3S

Starting read backup and flash new!
Now is: Friday, January 6, 2023 4:19:33 PM.
Flasher mode: BK7231T
Going to open port: COM3.
Serial port open!
Getting bus... (now, please do reboot by CEN or by power off/on)
Getting bus failed, will try again - 0/100!
Getting bus failed, will try again - 1/100!
Getting bus success!
Going to set baud rate setting (1500000)!
Going to start reading at offset 0x11000...
Reading 0x11000... failed with serial.BytesToRead 4075
Failed!

openshwprojects, 2023-01-06:
Does the problem happen only in "Backup and write" modes or in a separate modes as well?

I will try to prepare a build with increased time for read operation and we will see what happens.

Rafik89, 2023-01-06:
In my case both during backup and write. Also the write and read test in the advanced options do not allow to communicate properly with the MCU.

itsj4y, 2023-01-06:
same on backup (read) only
Starting read!
Now is: Friday, January 6, 2023 5:11:13 PM.
Flasher mode: BK7231T
Going to open port: COM3.
Serial port open!
Getting bus... (now, please do reboot by CEN or by power off/on)
Getting bus failed, will try again - 0/100!
Getting bus failed, will try again - 1/100!
Getting bus success!
Going to set baud rate setting (115200)!
Going to start reading at offset 0x11000...
Reading 0x11000... failed with serial.BytesToRead 4056
Failed! There was no result to save.

openshwprojects, 2023-01-06:
It works for me... I increased  a delay in UART read for slower computers...
[image]
Can you recheck?
https://github.com/openshwprojects/BK7231GUIFlashTool/releases/tag/v1.0.2

itsj4y, 2023-01-06:
1.0.2 doesn't give the countdown to do the reset

Starting read backup and flash new!
Now is: Friday, January 6, 2023 5:27:15 PM.
Flasher mode: BK7231T
Going to open port: COM3.
Serial port open!
Getting bus... (now, please do reboot by CEN or by power off/on)
Getting bus success!
Going to set baud rate setting (921600)!
Going to start reading at offset 0x11000...
Reading 0x11000... failed with serial.BytesToRead 4092 (expected 4111)
The beginning of buffer in UART contains 040EFF01E0FCF40610090000100100FF data.
Failed!

openshwprojects, 2023-01-06:
No @itsj4y , it's not the version thing. You won't get countdown to RESET if you do erase flash memory of BK but do not write. This is because in this case BK has no application and reboots constantly. 

Still... thanks for testing. I hoped that increasing the delay from 5 to 10 seconds will help, but it seems that issue is somewhere else.

Flash respond command is the following:
        byte[] cBuf = new byte[] { 0x04, 0x0e, 0xff, 0x01, 0xe0, 0xfc, 0xf4, (1 + 1 + (4 + 4 * 1024)) & 0xff,
                ((1 + 1 + (4 + 4 * 1024)) >> 8) & 0xff, (byte)CommandCode.FlashRead4K};

so your buffer content looks like it has a start of flash read command, but for some reasons it's cut in place? 

Can you try different baud rate, like 128000 and 256000 ? You can enter those numbers into the field (copy paste them, they are not on the list).

itsj4y, 2023-01-06:
[image]

itsj4y, 2023-01-06:
Ok so it worked after firmware write (no backup)

Rafik89, 2023-01-07:
When i do Backup:

```
Starting read!
Now is: sobota, 7 stycznia 2023 00:48:51.
Flasher mode: BK7231N
Going to open port: COM4.
Serial port open!
Getting bus... (now, please do reboot by CEN or by power off/on)
Getting bus failed, will try again - 0/100!
Getting bus failed, will try again - 1/100!
Getting bus failed, will try again - 2/100!
Getting bus failed, will try again - 3/100!
Getting bus success!
Going to set baud rate setting (921600)!
Will try to read device flash MID (for unprotect N):
Flash MID loaded: 15701C
Will now search for Flash def in out database...
Flash def found! For: 15701C
Flash information: mid: 15701C, icName: EN25QH16B, manufacturer: ESMT, szMem: 1000000, szSR: 1, cwUnp: 0, cwEnp: 7, cwMsk: 3C, sb: 2, lb: 4, cwdRd: 05-FF-FF-FF, cwdWr: 01-FF-FF-FF
Entering SetProtectState(True)...
sr: 7c
final sr: 7c
msk: 3c
cw: 0, sb: 2, lb: 4
bfd: 0
sr: 40
final sr: 40
msk: 3c
cw: 0, sb: 2, lb: 4
bfd: 0
SetProtectState(True) success!
Going to start reading at offset 0x00...
Reading 0x00... failed with serial.BytesToRead 4056 (expected 4111)
The beginning of buffer in UART contains 040EFF01E0FCF4061009000000000069 data.
Failed! There was no result to save.

```

```
Getting bus success!
Going to set baud rate setting (115200)!
Will try to read device flash MID (for unprotect N):
Flash MID loaded: 15701C
Will now search for Flash def in out database...
Flash def found! For: 15701C
Flash information: mid: 15701C, icName: EN25QH16B, manufacturer: ESMT, szMem: 1000000, szSR: 1, cwUnp: 0, cwEnp: 7, cwMsk: 3C, sb: 2, lb: 4, cwdRd: 05-FF-FF-FF, cwdWr: 01-FF-FF-FF
Entering SetProtectState(True)...
sr: 40
final sr: 40
msk: 3c
cw: 0, sb: 2, lb: 4
bfd: 0
SetProtectState(True) success!
Going to start reading at offset 0x00...
Reading 0x00... failed with serial.BytesToRead 4055 (expected 4111)
The beginning of buffer in UART contains 040EFF01E0FCF4061009000000000069 data.
Failed! There was no result to save.
```

```
Starting read!
Now is: sobota, 7 stycznia 2023 00:51:20.
Flasher mode: BK7231N
Going to open port: COM4.
Serial port open!
Getting bus... (now, please do reboot by CEN or by power off/on)
Getting bus failed, will try again - 0/100!
Getting bus failed, will try again - 1/100!
Getting bus failed, will try again - 2/100!
[... 123 lines cut ...]
Getting bus failed, will try again - 3/100!
Getting bus failed, will try again - 4/100!
Getting bus success!
Going to set baud rate setting (921600)!
Will try to read device flash MID (for unprotect N):
Flash MID loaded: 15701C
Will now search for Flash def in out database...
Flash def found! For: 15701C
Flash information: mid: 15701C, icName: EN25QH16B, manufacturer: ESMT, szMem: 1000000, szSR: 1, cwUnp: 0, cwEnp: 7, cwMsk: 3C, sb: 2, lb: 4, cwdRd: 05-FF-FF-FF, cwdWr: 01-FF-FF-FF
Entering SetProtectState(True)...
sr: 40
final sr: 40
msk: 3c
cw: 0, sb: 2, lb: 4
bfd: 0
SetProtectState(True) success!
Going to do erase, start 69632, sec count 2!
Erasing sector 69632... ok! Erasing sector 73728... ok! 
All selected sectors erased!
Going to start reading at offset 0x11000...
Reading 0x11000... failed with serial.BytesToRead 4092 (expected 4111)
The beginning of buffer in UART contains 040EFF01E0FCF40610090000100100FF data.
Failed! Read failed?
```
[[BK7231GUIFlashTool v1.0.2](https://github.com/openshwprojects/BK7231GUIFlashTool/releases/tag/v1.0.2)](https://github.com/openshwprojects/BK7231GUIFlashTool/commit/1de86ee441ce792ba1b0ff1c3b755ce820869595)

openshwprojects, 2023-01-07:
I have added a larger buffer, it was not needed for my machine, but maybe...
[image]
That 4096 seems suspicious, because it fails somewhere around that point, but again.... for some reasons it always work for me. Still, it's worthchecking.
Can you all check this release, please?
https://github.com/openshwprojects/BK7231GUIFlashTool/releases/tag/v1.0.3

Rafik89, 2023-01-07:
This was successfully done. You made it :)
With https://github.com/openshwprojects/BK7231GUIFlashTool/releases/tag/v1.0.3 works for me.

**baud rate 921600:**
```

Starting read!
Now is: sobota, 7 stycznia 2023 01:21:07.
Flasher mode: BK7231N
Going to open port: COM4.
Serial port open!
Getting bus... (now, please do reboot by CEN or by power off/on)
Getting bus failed, will try again - 0/100!
Getting bus failed, will try again - 1/100!
Getting bus failed, will try again - 2/100!
Getting bus success!
Going to set baud rate setting (921600)!
Will try to read device flash MID (for unprotect N):
Flash MID loaded: 15701C
Will now search for Flash def in out database...
Flash def found! For: 15701C
Flash information: mid: 15701C, icName: EN25QH16B, manufacturer: ESMT, szMem: 1000000, szSR: 1, cwUnp: 0, cwEnp: 7, cwMsk: 3C, sb: 2, lb: 4, cwdRd: 05-FF-FF-FF, cwdWr: 01-FF-FF-FF
Entering SetProtectState(True)...
sr: 40
final sr: 40
msk: 3c
cw: 0, sb: 2, lb: 4
bfd: 0
SetProtectState(True) success!
Going to start reading at offset 0x00...
Reading 0x00... Ok! Reading 0x1000... Ok! Reading 0x2000... Ok! Reading 0x3000... Ok! Reading 0x4000... Ok! Reading 0x5000... Ok! Reading 0x6000... Ok! Reading 0x7000... Ok! Reading 0x8000... Ok! Reading 0x9000... Ok! Reading 0xA000... Ok! Reading 0xB000... Ok! Reading 0xC000... Ok! Reading 0xD000... Ok! Reading 0xE000... Ok! Reading 0xF000... Ok! Reading 0x10000... Ok! Reading 0x11000... Ok! Reading 0x12000... Ok! Reading 0x13000... Ok! Reading 0x14000... Ok! Reading 0x15000... Ok! Reading 0x16000... Ok! Reading 0x17000... Ok! Reading 0x18000... Ok! Reading 0x19000... Ok! Reading 0x1A000... Ok! Reading 0x1B000... Ok! Reading 0x1C000... Ok! Reading 0x1D000... Ok! Reading 0x1E000... Ok! Reading 0x1F000... Ok! Reading 0x20000... Ok! Reading 0x21000... Ok! Reading 0x22000... Ok! Reading 0x23000... Ok! Reading 0x24000... Ok! Reading 0x25000... Ok! Reading 0x26000... Ok! Reading 0x27000... Ok! Reading 0x28000... Ok! Reading 0x29000... Ok! Reading 0x2A000... Ok! Reading 0x2B000... Ok! Reading 0x2C000... Ok! Reading 0x2D000... Ok! Reading 0x2E000... Ok! Reading 0x2F000... Ok! Reading 0x30000... Ok! Reading 0x31000... Ok! Reading 0x32000... Ok! Reading 0x33000... Ok! Reading 0x34000... Ok! Reading 0x35000... Ok! Reading 0x36000... Ok! Reading 0x37000... Ok! Reading 0x38000... Ok! Reading 0x39000... Ok! Reading 0x3A000... Ok! Reading 0x3B000... Ok! Reading 0x3C000... Ok! Reading 0x3D000... Ok! Reading 0x3E000... Ok! Reading 0x3F000... Ok! Reading 0x40000... Ok! Reading 0x41000... Ok! Reading 0x42000... Ok! Reading 0x43000... Ok! Reading 0x44000... Ok! Reading 0x45000... Ok! Reading 0x46000... Ok! Reading 0x47000... Ok! Reading 0x48000... Ok! Reading 0x49000... Ok! Reading 0x4A000... Ok! Reading 0x4B000... Ok! Reading 0x4C000... Ok! Reading 0x4D000... Ok! Reading 0x4E000... Ok! Reading 0x4F000... Ok! Reading 0x50000... Ok! Reading 0x51000... Ok! Reading 0x52000... Ok! Reading 0x53000... Ok! Reading 0x54000... Ok! Reading 0x55000... Ok! Reading 0x56000... Ok! Reading 0x57000... Ok! Reading 0x58000... Ok! Reading 0x59000... Ok! Reading 0x5A000... Ok! Reading 0x5B000... Ok! Reading 0x5C000... Ok! Reading 0x5D000... Ok! Reading 0x5E000... Ok! Reading 0x5F000... Ok! Reading 0x60000... Ok! Reading 0x61000... Ok! Reading 0x62000... Ok! Reading 0x63000... Ok! Reading 0x64000... Ok! Reading 0x65000... Ok! Reading 0x66000... Ok! Reading 0x67000... Ok! Reading 0x68000... Ok! Reading 0x69000... Ok! Reading 0x6A000... Ok! Reading 0x6B000... Ok! Reading 0x6C000... Ok! Reading 0x6D000... Ok! Reading 0x6E000... Ok! Reading 0x6F000... Ok! Reading 0x70000... Ok! Reading 0x71000... Ok! Reading 0x72000... Ok! Reading 0x73000... Ok! Reading 0x74000... Ok! Reading 0x75000... Ok! Reading 0x76000... Ok! Reading 0x77000... Ok! Reading 0x78000... Ok! Reading 0x79000... Ok! Reading 0x7A000... Ok! Reading 0x7B000... Ok! Reading 0x7C000... Ok! Reading 0x7D000... Ok! Reading 0x7E000... Ok! Reading 0x7F000... Ok! Reading 0x80000... Ok! Reading 0x81000... Ok! Reading 0x82000... Ok! Reading 0x83000... Ok! Reading 0x84000... Ok! Reading 0x85000... Ok! Reading 0x86000... Ok! Reading 0x87000... Ok! Reading 0x88000... Ok! Reading 0x89000... Ok! Reading 0x8A000... Ok! Reading 0x8B000... Ok! Reading 0x8C000... Ok! Reading 0x8D000... Ok! Reading 0x8E000... Ok! Reading 0x8F000... Ok! Reading 0x90000... Ok! Reading 0x91000... Ok! Reading 0x92000... Ok! Reading 0x93000... Ok! Reading 0x94000... Ok! Reading 0x95000... Ok! Reading 0x96000... Ok! Reading 0x97000... Ok! Reading 0x98000... Ok! Reading 0x99000... Ok! Reading 0x9A000... Ok! Reading 0x9B000... Ok! Reading 0x9C000... Ok! Reading 0x9D000... Ok! Reading 0x9E000... Ok! Reading 0x9F000... Ok! Reading 0xA0000... Ok! Reading 0xA1000... Ok! Reading 0xA2000... Ok! Reading 0xA3000... Ok! Reading 0xA4000... Ok! Reading 0xA5000... Ok! Reading 0xA6000... Ok! Reading 0xA7000... Ok! Reading 0xA8000... Ok! Reading 0xA9000... Ok! Reading 0xAA000... Ok! Reading 0xAB000... Ok! Reading 0xAC000... Ok! Reading 0xAD000... Ok! Reading 0xAE000... Ok! Reading 0xAF000... Ok! Reading 0xB0000... Ok! Reading 0xB1000... Ok! Reading 0xB2000... Ok! Reading 0xB3000... Ok! Reading 0xB4000... Ok! Reading 0xB5000... Ok! Reading 0xB6000... Ok! Reading 0xB7000... Ok! Reading 0xB8000... Ok! Reading 0xB9000... Ok! Reading 0xBA000... Ok! Reading 0xBB000... Ok! Reading 0xBC000... Ok! Reading 0xBD000... Ok! Reading 0xBE000... Ok! Reading 0xBF000... Ok! Reading 0xC0000... Ok! Reading 0xC1000... Ok! Reading 0xC20 [... cut ...]
Erasing sector 69632... ok! Erasing sector 73728... ok! 
All selected sectors erased!
Writing sector 69632... ok! Writing sector 73728... ok! Starting CRC check for 2 sectors, starting at offset 11000
CRC matches 0x4998ACF8!
Going to start reading at offset 0x11000...
Reading 0x11000... Ok! Reading 0x12000... Ok! 
Basic read operation finished, but now it's time to verify...
Starting CRC check for 2 sectors, starting at offset 11000
CRC matches 0x4998ACF8!
All read!
Loaded total 0x2000 bytes 
Check passed! Loaded data was the same as written!**

openshwprojects, 2023-01-07:
Thank you @Rafik89 for testing. It's a good news. I am still super surprised that it worked for me with 4096 buffer, it's very strange. 
Still, I would like everyone else here to test it as well, because I am not yet sure that it's really solved.

az-src, 2023-01-07:
I checked WB3S reading, everything works. I will try to restore the original firmware of the device. I will help you test
Flasher mode: BK7231T
Going to open port: COM6.
Serial port open!
Getting bus... (now, please do reboot by CEN or by power off/on)
Getting bus failed, will try again - 0/100!
Getting bus failed, will try again - 1/100!
Getting bus failed, will try again - 2/100!
Getting bus success!
Going to set baud rate setting (115200)!
Going to start reading at offset 0x11000...
.........
Reading 0x20F000... Ok! Reading 0x210000... Ok! 
Basic read operation finished, but now it's time to verify...
Starting CRC check for 512 sectors, starting at offset 11000
CRC matches 0x35D1A1EC!
All read!
Loaded total 0x200000 bytes 
Wrote 2097152 to readResult_BK7231T_2023-07-1--05-28-41.bin

openshwprojects, 2023-01-07:
Do not try to restore the original firmware without me adding an option for that first! You may break your bootloader.

az-src, 2023-01-07:
OK got it

veivari, 2023-01-07:
Hello, I have problem flashing CB2S.
I'm using ftdi ft232rl and external power suplier.
I tried all baudrates mentioned here and problem was allways same:

Getting bus success!
Going to set baud rate setting (115200)!
Will try to read device flash MID (for unprotect N):
Flash MID loaded: 1560EB
Will now search for Flash def in out database...
Flash def found! For: 1560EB
Flash information: mid: 1560EB, icName: TH25Q_16HB, manufacturer: TH, szMem: 1000000, szSR: 2, cwUnp: 0, cwEnp: 7, cwMsk: 407C, sb: 2, lb: 5, cwdRd: 05-35-FF-FF, cwdWr: 01-FF-FF-FF
Entering SetProtectState(True)...
sr: 0
sr: 0
final sr: 0
msk: 407c
cw: 0, sb: 2, lb: 5
bfd: 0
SetProtectState(True) success!
Going to start reading at offset 0x00...
Reading 0x00... failed with serial.BytesToRead 4095 (expected 4111)
The beginning of buffer in UART contains 040EFF01E0FCF4061009000000000069 data.
Failed! There was no result to save.

openshwprojects, 2023-01-07:
@veivari  which version?

veivari, 2023-01-07:
@openshwprojects 1.0.3

klmi, 2023-01-12:
Hi @openshwprojects,
I'm  facing the same issue as @veivari but using version 1.0.4
I tried a couple of baudrates but no success.
Using CH34x as USB to TTL guess it's from AZ Delivey.

dawosch, 2023-01-26:
I also get the same problem:
```
Starting read backup and flash new!
Now is: Donnerstag, 26. Januar 2023 23:50:05.
Flasher mode: BK7231N
Going to open port: COM4.
Serial port open!
Getting bus... (now, please do reboot by CEN or by power off/on)
Getting bus failed, will try again - 0/100!
Getting bus failed, will try again - 1/100!
Getting bus success!
Going to set baud rate setting (921600)!
Will try to read device flash MID (for unprotect N):
Flash MID loaded: 1560EB
Will now search for Flash def in out database...
Flash def found! For: 1560EB
Flash information: mid: 1560EB, icName: TH25Q_16HB, manufacturer: TH, szMem: 1000000, szSR: 2, cwUnp: 0, cwEnp: 7, cwMsk: 407C, sb: 2, lb: 5, cwdRd: 05-35-FF-FF, cwdWr: 01-FF-FF-FF
Entering SetProtectState(True)...
sr: 80
sr: 680
final sr: 680
msk: 407c
cw: 0, sb: 2, lb: 5
bfd: 0
SetProtectState(True) success!
Going to start reading at offset 0x00...
Reading 0x00... Ok! Reading 0x1000... Ok! Reading 0x2000... Ok! Reading 0x3000... Ok! Reading 0x4000... Ok! Reading 0x5000... Ok! Reading 0x6000... Ok! Reading 0x7000... Ok! Reading 0x8000... Ok! Reading 0x9000... Ok! Reading 0xA000... Ok! Reading 0xB000... Ok! Reading 0xC000... Ok! Reading 0xD000... Ok! Reading 0xE000... Ok! Reading 0xF000... Ok! Reading 0x10000... Ok! Reading 0x11000... Ok! Reading 0x12000... Ok! Reading 0x13000... Ok! Reading 0x14000... Ok! Reading 0x15000... Ok! Reading 0x16000... Ok! Reading 0x17000... Ok! Reading 0x18000... Ok! Reading 0x19000... Ok! Reading 0x1A000... Ok! Reading 0x1B000... Ok! Reading 0x1C000... Ok! Reading 0x1D000... Ok! Reading 0x1E000... Ok! Reading 0x1F000... Ok! Reading 0x20000... Ok! Reading 0x21000... Ok! Reading 0x22000... Ok! Reading 0x23000... Ok! Reading 0x24000... Ok! Reading 0x25000... Ok! Reading 0x26000... Ok! Reading 0x27000... Ok! Reading 0x28000... Ok! Reading 0x29000... Ok! Reading 0x2A000... Ok! Reading 0x2B000... Ok! Reading 0x2C000... Ok! Reading 0x2D000... Ok! Reading 0x2E000... Ok! Reading 0x2F000... Ok! Reading 0x30000... Ok! Reading 0x31000... Ok! Reading 0x32000... Ok! Reading 0x33000... Ok! Reading 0x34000... Ok! Reading 0x35000... Ok! Reading 0x36000... Ok! Reading 0x37000... Ok! Reading 0x38000... Ok! Reading 0x39000... Ok! Reading 0x3A000... Ok! Reading 0x3B000... Ok! Reading 0x3C000... Ok! Reading 0x3D000... Ok! Reading 0x3E000... Ok! Reading 0x3F000... Ok! Reading 0x40000... Ok! Reading 0x41000... Ok! Reading 0x42000... Ok! Reading 0x43000... Ok! Reading 0x44000... Ok! Reading 0x45000... Ok! Reading 0x46000... Ok! Reading 0x47000... Ok! Reading 0x48000... Ok! Reading 0x49000... Ok! Reading 0x4A000... Ok! Reading 0x4B000... Ok! Reading 0x4C000... Ok! Reading 0x4D000... Ok! Reading 0x4E000... Ok! Reading 0x4F000... Ok! Reading 0x50000... Ok! Reading 0x51000... Ok! Reading 0x52000... Ok! Reading 0x53000... Ok! Reading 0x54000... Ok! Reading 0x55000... Ok! Reading 0x56000... Ok! Reading 0x57000... Ok! Reading 0x58000... Ok! Reading 0x59000... Ok! Reading 0x5A000... Ok! Reading 0x5B000... Ok! Reading 0x5C000... Ok! Reading 0x5D000... Ok! Reading 0x5E000... Ok! Reading 0x5F000... Ok! Reading 0x60000... Ok! Reading 0x61000... Ok! Reading 0x62000... Ok! Reading 0x63000... Ok! Reading 0x64000... Ok! Reading 0x65000... Ok! Reading 0x66000... Ok! Reading 0x67000... Ok! Reading 0x68000... Ok! Reading 0x69000... Ok! Reading 0x6A000... Ok! Reading 0x6B000... Ok! Reading 0x6C000... Ok! Reading 0x6D000... Ok! Reading 0x6E000... Ok! Reading 0x6F000... Ok! Reading 0x70000... Ok! Reading 0x71000... Ok! Reading 0x72000... Ok! Reading 0x73000... Ok! Reading 0x74000... Ok! Reading 0x75000... Ok! Reading 0x76000... Ok! Reading 0x77000... Ok! Reading 0x78000... Ok! Reading 0x79000... Ok! Reading 0x7A000... Ok! Reading 0x7B000... Ok! Reading 0x7C000... Ok! Reading 0x7D000... Ok! Reading 0x7E000... Ok! Reading 0x7F000... Ok! Reading 0x80000... Ok! Reading 0x81000... Ok! Reading 0x82000... Ok! Reading 0x83000... Ok! Reading 0x84000... Ok! Reading 0x85000... Ok! Reading 0x86000... Ok! Reading 0x87000... Ok! Reading 0x88000... Ok! Reading 0x89000... Ok! Reading 0x8A000... Ok! Reading 0x8B000... Ok! Reading 0x8C000... Ok! Reading 0x8D000... Ok! Reading 0x8E000... Ok! Reading 0x8F000... Ok! Reading 0x90000... Ok! Reading 0x91000... Ok! Reading 0x92000... Ok! Reading 0x93000... Ok! Reading 0x94000... Ok! Reading 0x95000... Ok! Reading 0x96000... Ok! Reading 0x97000... Ok! Reading 0x98000... Ok! Reading 0x99000... Ok! Reading 0x9A000... Ok! Reading 0x9B000... Ok! Reading 0x9C000... Ok! Reading 0x9D000... Ok! Reading 0x9E000... Ok! Reading 0x9F000... Ok! Reading 0xA0000... Ok! Reading 0xA1000... Ok! Reading 0xA2000... failed with serial.BytesToRead 0 (expected 4111)
The beginning of buffer in UART contains  data.
Failed! 
```

I'm using this USB to UART device:
https://image.made-in-china.com/2f0j00levUBmHLfGos/FT232r-USB-Uart-Nano.jpg

openshwprojects, 2023-01-27:
@dawosch isn't it a battery-powered device where MCU turns off the wifi module? Can you show device photo?

dawosch, 2023-01-27:
@Rafik89 
Yes it is. 
Ignore the wirering. I tested a lot.
My approaches based on the connection direct of the chip. 

[image]

openshwprojects, 2023-01-27:
Sorry but your problem is not related to the flasher. Basically, this device  has TuyaMCU and TuyaMCU might interfere with flashing. You can either cut the rx/tx tracks or desolder the MCU temporarily. 
You can read and ask more here:
https://www.elektroda.com/rtvforum/topic3914412.html

dawosch, 2023-01-27:
Ok, thanks for the info. 
But one more question/idea. 
At the current highest baud rate i can read or write round about 80% of the firmware before it interrupts. 

Would it be possible to increase the baud rate a little bit more to write the firmware complete?
Or reduce the firmware size for faster writing?

Thanks

openshwprojects, 2023-01-29:
If you are asking in relation to TuyaMCU devices, then it's not practical. The correct approach is just to power the device correctly (avoid MCU interruption) and, if required, remove temporarily the MCU or cut traces.

maciejla, 2023-02-07:
I have the same issue with BK7231N - it's from a smart plug, so I don't think there is the mentioned above TuyaMCU on board (attaching pictures just in case I can't see it). I have tried different baudrates
[image]
[image]
I'm using 1.0.4 version of the flasher
Here are my logs:
`Test read/write pattern`
```
Starting read-write test!
Now is: Tuesday, February 7, 2023 3:51:32 PM.
Flasher mode: BK7231N
Going to open port: /dev/ttyUSB0.
Serial port open!
Getting bus... (now, please do reboot by CEN or by power off/on)
Getting bus failed, will try again - 0/100!
Getting bus failed, will try again - 1/100!
Getting bus success!
Going to set baud rate setting (921600)!
Will try to read device flash MID (for unprotect N):
Flash MID loaded: 1560EB
Will now search for Flash def in out database...
Flash def found! For: 1560EB
Flash information: mid: 1560EB, icName: TH25Q_16HB, manufacturer: TH, szMem: 1000000, szSR: 2, cwUnp: 0, cwEnp: 7, cwMsk: 407C, sb: 2, lb: 5, cwdRd: 05-35-FF-FF, cwdWr: 01-FF-FF-FF
Entering SetProtectState(True)...
sr: 0
sr: 0
final sr: 0
msk: 407c
cw: 0, sb: 2, lb: 5
bfd: 0
SetProtectState(True) success!
Going to do erase, start 69632, sec count 2!
Erasing sector 69632... ok! Erasing sector 73728... ok! 
All selected sectors erased!
Going to start reading at offset 0x11000...
Reading 0x11000... failed with serial.BytesToRead 4095 (expected 4111)
The beginning of buffer in UART contains 040EFF01E0FCF40610090000102100FF data.
Failed! Read failed?
```
`Test write pattern`
```

Starting write test!
Now is: Tuesday, February 7, 2023 3:53:35 PM.
Flasher mode: BK7231N
Going to open port: /dev/ttyUSB0.
Serial port open!
Getting bus... (now, please do reboot by CEN or by power off/on)
Getting bus failed, will try again - 0/100!
Getting bus failed, will try again - 1/100!
Getting bus failed, will try again - 2/100!
Getting bus success!
Going to set baud rate setting (921600)!
Will try to read device flash MID (for unprotect N):
Flash MID loaded: 1560EB
Will now search for Flash def in out database...
Flash def found! For: 1560EB
Flash information: mid: 1560EB, icName: TH25Q_16HB, manufacturer: TH, szMem: 1000000, szSR: 2, cwUnp: 0, cwEnp: 7, cwMsk: 407C, sb: 2, lb: 5, cwdRd: 05-35-FF-FF, cwdWr: 01-FF-FF-FF
Entering SetProtectState(True)...
sr: 0
sr: 0
final sr: 0
msk: 407c
cw: 0, sb: 2, lb: 5
bfd: 0
SetProtectState(True) success!
Erasing sector 0x11000... ok! Erasing sector 0x12000... ok! Erasing sector 0x13000... ok! Erasing sector 0x14000... ok! Erasing sector 0x15000... ok! 
All selected sectors erased!
failed with serial.BytesToRead 0 (expected 15)
The beginning of buffer in UART contains  data.
Writing sector 69632... Writing sector 69632 failed!
[... 64 lines cut ...]
Preparing to write data file to chip - resetting bus and baud...
Getting bus... (now, please do reboot by CEN or by power off/on)
Getting bus success!
Going to set baud rate setting (921600)!
Will try to read device flash MID (for unprotect N):
Flash MID loaded: 1560EB
Will now search for Flash def in out database...
Flash def found! For: 1560EB
Flash information: mid: 1560EB, icName: TH25Q_16HB, manufacturer: TH, szMem: 1000000, szSR: 2, cwUnp: 0, cwEnp: 7, cwMsk: 407C, sb: 2, lb: 5, cwdRd: 05-35-FF-FF, cwdWr: 01-FF-FF-FF
Entering SetProtectState(True)...
sr: 0
sr: 0
final sr: 0
msk: 407c
cw: 0, sb: 2, lb: 5
bfd: 0
SetProtectState(True) success!
Going to do erase, start 0, sec count 298!
Erasing sector 0... ok! Erasing sector 4096... ok! Erasing sector 8192... ok! Erasing sector 12288... ok! Erasing sector 16384... ok! Erasing sector 20480... ok! Erasing sector 24576... ok! Erasing sector 28672... ok! Erasing sector 32768... ok! Erasing sector 36864... ok! Erasing sector 40960... ok! Erasing sector 45056... ok! Erasing sector 49152... ok! Erasing sector 53248... ok! Erasing sector 57344... ok! Erasing sector 61440... ok! Erasing sector 65536... ok! Erasing sector 69632... ok! Erasing sector 73728... ok! Erasing sector 77824... ok! Erasing sector 81920... ok! Erasing sector 86016... ok! Erasing sector 90112... ok! Erasing sector 94208... ok! Erasing sector 98304... ok! Erasing sector 102400... ok! Erasing sector 106496... ok! Erasing sector 110592... ok! Erasing sector 114688... ok! Erasing sector 118784... ok! Erasing sector 122880... ok! Erasing sector 126976... ok! Erasing sector 131072... ok! Erasing sector 135168... ok! Erasing sector 139264... ok! Erasing sector 143360... ok! Erasing sector 147456... ok! Erasing sector 151552... ok! Erasing sector 155648... ok! Erasing sector 159744... ok! Erasing sector 163840... ok! Erasing sector 167936... ok! Erasing sector 172032... ok! Erasing sector 176128... ok! Erasing sector 180224... ok! Erasing sector 184320... ok! Erasing sector 188416... ok! Erasing sector 192512... ok! Erasing sector 196608... ok! Erasing sector 200704... ok! Erasing sector 204800... ok! Erasing sector 208896... ok! Erasing sector 212992... ok! Erasing sector 217088... ok! Erasing sector 221184... ok! Erasing sector 225280... ok! Erasing sector 229376... ok! Erasing sector 233472... ok! Erasing sector 237568... ok! Erasing sector 241664... ok! Erasing sector 245760... ok! Erasing sector 249856... ok! Erasing sector 253952... ok! Erasing sector 258048... ok! Erasing sector 262144... ok! Erasing sector 266240... ok! Erasing sector 270336... ok! Erasing sector 274432... ok! Erasing sector 278528... ok! Erasing sector 282624... ok! Erasing sector 286720... ok! Erasing sector 290816... ok! Erasing sector 29491 [... cut ...] ctor 1183744... ok! Erasing sector 1187840... ok! Erasing sector 1191936... ok! Erasing sector 1196032... ok! Erasing sector 1200128... ok! Erasing sector 1204224... ok! Erasing sector 1208320... ok! Erasing sector 1212416... ok! Erasing sector 1216512... ok! 
All selected sectors erased!
failed with serial.BytesToRead 0 (expected 15)
The beginning of buffer in UART contains 040EFF01E0FCF40600070000000000 data.
Writing sector 0x00... Writing sector 0 failed!
Writing file data to chip failed.
```

Crakett, 2023-03-10:
Hello, I also had this random write error. I increased the TimeOut (0.25f instead of 0.05f) at line 387 of the BK7231Flasher.cs file (from V1.06) and the write was successful.

Dexruus, 2023-03-11:
Hello,

iam having an problem with an BK7231N on an aliexpress relay board.

Firmware read was a success at first, the write is the problem (thats the reason why i dont do a backup in this try)
it randomly stops either at 0x4000 or 0x1000. At higher baudrate there was no connectivity at all 115200 did following.

`Starting flash new (no backup)!
Now is: Samstag, 11. März 2023 22:39:00.
Flasher mode: BK7231N
Going to open port: COM3.
Serial port open!
Getting bus... (now, please do reboot by CEN or by power off/on)
Getting bus failed, will try again - 0/100!
Getting bus failed, will try again - 1/100!
Getting bus success!
Going to set baud rate setting (115200)!
Will try to read device flash MID (for unprotect N):
Flash MID loaded: 1560EB
Will now search for Flash def in out database...
Flash def found! For: 1560EB
Flash information: mid: 1560EB, icName: TH25Q_16HB, manufacturer: TH, szMem: 1000000, szSR: 2, cwUnp: 0, cwEnp: 7, cwMsk: 407C, sb: 2, lb: 5, cwdRd: 05-35-FF-FF, cwdWr: 01-FF-FF-FF
Entering SetProtectState(True)...
sr: 0
sr: 0
final sr: 0
msk: 407c
cw: 0, sb: 2, lb: 5
bfd: 0
SetProtectState(True) success!
Reading file firmwares/OpenBK7231N_QIO_1.15.450.bin...
Loaded 1220464 bytes from firmwares/OpenBK7231N_QIO_1.15.450.bin...
Preparing to write data file to chip - resetting bus and baud...
Getting bus... (now, please do reboot by CEN or by power off/on)
Getting bus success!
Going to set baud rate setting (115200)!
Will try to read device flash MID (for unprotect N):
Flash MID loaded: 1560EB
Will now search for Flash def in out database...
Flash def found! For: 1560EB
Flash information: mid: 1560EB, icName: TH25Q_16HB, manufacturer: TH, szMem: 1000000, szSR: 2, cwUnp: 0, cwEnp: 7, cwMsk: 407C, sb: 2, lb: 5, cwdRd: 05-35-FF-FF, cwdWr: 01-FF-FF-FF
Entering SetProtectState(True)...
sr: 0
sr: 0
final sr: 0
msk: 407c
cw: 0, sb: 2, lb: 5
bfd: 0
SetProtectState(True) success!
Going to do erase, start 0, sec count 298!
Erasing sector 0... ok! Erasing sector 4096... ok! Erasing sector 8192... ok! Erasing sector 12288... ok! Erasing sector 16384... ok! Erasing sector 20480... ok! Erasing sector 24576... ok! Erasing sector 28672... ok! Erasing sector 32768... ok! Erasing sector 36864... ok! Erasing sector 40960... ok! Erasing sector 45056... ok! Erasing sector 49152... ok! Erasing sector 53248... ok! Erasing sector 57344... ok! Erasing sector 61440... ok! Erasing sector 65536... ok! Erasing sector 69632... ok! Erasing sector 73728... ok! Erasing sector 77824... ok! Erasing sector 81920... ok! Erasing sector 86016... ok! Erasing sector 90112... ok! Erasing sector 94208... ok! Erasing sector 98304... ok! Erasing sector 102400... ok! Erasing sector 106496... ok! Erasing sector 110592... ok! Erasing sector 114688... ok! Erasing sector 118784... ok! Erasing sector 122880... ok! Erasing sector 126976... ok! Erasing sector 131072... ok! Erasing sector 135168... ok! Erasing sector 139264... ok! Erasing sector 143360... ok! Erasing sector 147456... ok! Erasing sector 151552... ok! Erasing sector 155648... ok! Erasing sector 159744... ok! Erasing sector 163840... ok! Erasing sector 167936... ok! Erasing sector 172032... ok! Erasing sector 176128... ok! Erasing sector 180224... ok! Erasing sector 184320... ok! Erasing sector 188416... ok! Erasing sector 192512... ok! Erasing sector 196608... ok! Erasing sector 200704... ok! Erasing sector 204800... ok! Erasing sector 208896... ok! Erasing sector 212992... ok! Erasing sector 217088... ok! Erasing sector 221184... ok! Erasing sector 225280... ok! Erasing sector 229376... ok! Erasing sector 233472... ok! Erasing sector 237568... ok! Erasing sector 241664... ok! Erasing sector 245760... ok! Erasing sector 249856... ok! Erasing sector 253952... ok! Erasing sector 258048... ok! Erasing sector 262144... ok! Erasing sector 266240... ok! Erasing sector 270336... ok! Erasing sector 274432... ok! Erasing sector 278528... ok! Erasing sector 282624... ok! Erasing sector 286720... ok! Erasing sector 290816... ok! Erasing sector 294912... ok! Erasing sector 299008... ok! Erasing sector 303104... ok! Erasing sector 307200... ok! Erasing sector 311296... ok! Erasing sector 315392... ok! Erasing sector 319488... ok! Erasing sector 323584... ok! Erasing sector 327680... ok! Erasing sector 331776... ok! Erasing sector 335872... ok! Erasing sector 339968... ok! Erasing sector 344064... ok! Erasing sector 348160... ok! Erasing sector 352256... ok! Erasing sector 356352... ok! Erasing sector 360448... ok! Erasing sector 364544... ok! Erasing sector 368640... ok! Erasing sector 372736... ok! Erasing sector 376832... ok! Erasing sector 380928... ok! Erasing sector 385024... ok! Erasing sector 389120... ok! Erasing sector 393216... ok! Erasing sector 397312... ok! Erasing sector 401408... ok! Erasing sector 405504... ok! Erasing sector 409600... ok! Erasing sector 413696... ok! Erasing sector 417792... ok! Erasing sector 421888... ok! Erasing sector 425984... ok! Erasing sector 430080... ok! Erasing sector 434176... ok! Erasing sector 438272... ok! Erasing sector 442368... ok! Erasing sector 446464... ok! Erasing sector 450560... ok! Erasing sector 454656... ok! Erasing sector 458752... ok! Erasing sector 462848... ok! Erasing sector 466944... ok! Erasing sector 471040... ok! Erasing sector 475136... ok! Erasing sector 479232... ok! Erasing sector 483328... ok! Erasing sector 487424... ok! Erasing sector 491520... ok! Erasing sector 495616... ok! Erasing sector 499712... ok! Erasing se [... cut ...] sector 1200128... ok! Erasing sector 1204224... ok! Erasing sector 1208320... ok! Erasing sector 1212416... ok! Erasing sector 1216512... ok! 
All selected sectors erased!
Writing sector 0x00... ok! Writing sector 0x1000... ok! Writing sector 0x2000... ok! Writing sector 0x3000... ok! failed with serial.BytesToRead 14 (expected 15)
The beginning of buffer in UART contains 040EFF01E0FCF40600070000400000 data.
Writing sector 0x4000... Writing sector 16384 failed!
Writing file data to chip failed.`

Crakett, 2023-03-12:
Hi Dexruus,

did you try to modify the timeout as I said in my previous message ?

openshwprojects, 2023-05-09:
I don't know what is wrong. It seems to work reliable for me. @thinksilicon  how did you compile it on linux?

Btw, I have a major update coming for this tool. It will be possible to extract Tuya config very easily and also it will be possible to configure OBK in place, from the flasher access. It will be released in few days.

thinksilicon, 2023-05-09:
Hey,
I built it with xbuild, which did give a deprecation warning, but still built fine. Reading code didn't succeed at all. It did make a connection to the bus after I toggled CEN, however it did not read any bytes.
Writing started with erasing like the previous posts in this issue and then errored out at various times during the erase. Maybe it's just some oddity with Mono + serial ports under Linux.
```
xbuild --help

>>>> xbuild tool is deprecated and will be removed in future updates, use msbuild instead <<<<

XBuild Engine Version 14.0
Mono, Version 6.12.0.122
Copyright (C) 2005-2013 Various Mono authors
```

fflores97, 2023-05-11:
Hey, just wanted to say that I was having a similar issue on linux but it worked fine on windows. As you said, it probably has to do with the interaction with serial ports over Mono. Hopefully they can find a fix!

LazyGatto, 2023-06-30:
> It works for me... I increased a delay in UART read for slower computers...
> [image]
> Can you recheck?
> https://github.com/openshwprojects/BK7231GUIFlashTool/releases/tag/v1.0.2

Great! Thanks for this hint! =)
Increasing delays worked for me too. Tried in Linux (Ubuntu) ARM on Parallels VM on Mac M1.

OlliKantola, 2023-07-12:
> > It works for me... I increased a delay in UART read for slower computers...
> > [image]
> > Can you recheck?
> > https://github.com/openshwprojects/BK7231GUIFlashTool/releases/tag/v1.0.2
> 
> Great! Thanks for this hint! =) Increasing UART timeouts worked for me too (Set to 5 first and second). Tried in Linux (Ubuntu) ARM on Parallels VM on Mac M1. [image]

Thanks, this helped. No issues on my Windows 10 laptop. But on my Ubuntu 22.04 workstation (Ryzen 5700G)  I faced the same issue. By changing two of the topmost UART timings from 1s to 5s everything started working.

openshwprojects, 2023-07-12:
Thank you for testing. Now 5s delays in those two fields are default values.
I've made a release - BK7231 GUI Flash Tool 1.1.1

hg42, 2023-07-18:
with a separated CB2S module, I got the sequence

```
SetProtectState(True) success!
Going to start reading at offset 0x00...
Reading 0x00... failed with serial.BytesToRead 4095 (expected 4111)
The beginning of buffer in UART contains 040EFF01E0FCF4061009000000200069 data.
Failed! There was no result to save.
```

on a Debian "testing" system on a Ryzen 5 4650G with version 1.1.1 (so with the two defaults of 5, I checked them).
With 9 sec in both entries it only waits longer after the
`Going to start reading at offset 0x00...`
line.

No problem on a Windows laptop with the same hardware.
My serial adapter is nearly exactly the same like the one above.

May be the higher timeout wasn't the reason it worked but only increased the chances.

vinibali, 2023-12-08:
Might relates to #19 
> I had the same problem on Arch Linux. Write was working fine, but it stopped at the backup.
> failed with serial.BytesToRead 4095 (expected 4111)
> It might worth a try to use this:
> https://github.com/OpenBekenIOT/hid_download_py
> Best regards