openshwprojects/BK7231GUIFlashTool #11: Produced dump is "swapped" i.e. bootloader is at the end of flash

Yamacore, 2023-04-19:
Hello there, i just did an dump as explained here: "https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/359"
And it seems the bootloader is at the end of flash.

openshwprojects, 2023-04-19:
Thank you for reporting the problem, how did you managed to get swapped binary? We fixed it long, long time ago... which tool version do you have? Can you post a screenshot?

Yamacore, 2023-04-19:
[image]
I used the "bk7231flasher_1.0.7.zip" the dumped file is uploaded in the cloudcutter issue i linked, maybe an error on my side?

openshwprojects, 2023-04-20:
Thank you, it seems that one of the buttons was not updated. Everyone must be using the auto read and write so no one noticed yet.
Thanks!

closed 2023-04-20 by openshwprojects