openshwprojects/BK7231GUIFlashTool #13: Flasher crashes after reset

jvelder, 2023-05-09:
Hi, 
following the recommended steps after reseting the beken the flasher repeatedly crashes with the following message:
`Going to open port: COM3.
Serial port open!
Getting bus... (now, please do reboot by CEN or by power off/on)
Getting bus failed, will try again - 0/100!
Getting bus failed, will try again - 1/100!
Getting bus failed, will try again - 2/100!
Got exception: System.UnauthorizedAccessException: Der Zugriff auf den Anschluss wurde verweigert.
   bei System.IO.Ports.InternalResources.WinIOError(Int32 errorCode, String str)
   bei System.IO.Ports.SerialStream.get_BytesToRead()
   bei System.IO.Ports.SerialPort.get_BytesToRead()
   bei BK7231Flasher.BK7231Flasher.Start_Cmd(Byte[] txbuf, Int32 rxLen, Single timeout)!
Exception caught: System.InvalidOperationException: Der Anschluss ist geschlossen.
   bei System.IO.Ports.SerialPort.get_BytesToRead()
   bei BK7231Flasher.BK7231Flasher.consumePending()
   bei BK7231Flasher.BK7231Flasher.Start_Cmd(Byte[] txbuf, Int32 rxLen, Single timeout)
   bei BK7231Flasher.BK7231Flasher.linkCheck()
   bei BK7231Flasher.BK7231Flasher.getBus()
   bei BK7231Flasher.BK7231Flasher.doGetBusAndSetBaudRate()
   bei BK7231Flasher.BK7231Flasher.doGenericSetup()
   bei BK7231Flasher.BK7231Flasher.doReadAndWriteInternal(Int32 startSector, Int32 sectors, String sourceFileName, Boolean bSkipWrite)
   bei BK7231Flasher.BK7231Flasher.doReadAndWrite(Int32 startSector, Int32 sectors, String sourceFileName, Boolean bSkipWrite)
`
Translating the German parts:
Der Zugriff auf den Anschluss wurde verweigert. = Access to the COM-Port was denied.
Der Anschluss ist geschlossen. = COM-Port closed.

I'm using Windows 10/64bit with Dot Net Framework 3.5 (also tried the current Mono framework). Both as a regular user as well as running the the flash tool as an admin.
I used an AZDelivery FT232RL.

openshwprojects, 2023-05-09:
Hello, does the previous version also crash?

Do you have Cura or some other app that takes control over UART port and blocks other applications from using it?

jvelder, 2023-05-09:
Hi,
went through all the installed apps and there's nothing that uses the UART-port (pretty much a barebones Windows that I only use for stuff like this on an old laptop). Also tried 1.0.8 & 1.0.7, same error.
Wiring seems to be OK as well, also tried a different USB-cable.

I thought I got while trying out things: Could this also be a problem with the adapter not delivering enough power to the Beken?

openshwprojects, 2023-05-09:
you should use external power supply, what is your current setup?

jvelder, 2023-05-12:
The lab PSU fixed it, worked fine the first try with it connected (using version 1.0.9).

closed 2023-05-12 by jvelder