openshwprojects/BK7231GUIFlashTool #14: Failed to set baud rate

citarone, 2023-05-16:
I always bekome this error i tried evrithing 

Starting flash new (no backup)!
Now is: Dienstag, 16. Mai 2023 10:36:26.
Flasher mode: BK7231N
Going to open port: COM6.
Serial port open!
Getting bus... (now, please do reboot by CEN or by power off/on)
Getting bus failed, will try again - 0/100!
Getting bus failed, will try again - 1/100!
Getting bus failed, will try again - 2/100!
Getting bus failed, will try again - 3/100!
Getting bus failed, will try again - 4/100!
Getting bus failed, will try again - 5/100!
Getting bus failed, will try again - 6/100!
Getting bus failed, will try again - 7/100!
Getting bus success!
Going to set baud rate setting (1500000)!
Failed to set baud rate!

openshwprojects, 2023-05-21:
How is your setup? Do you have long wires - try shorting them? Which USB to uart converer? Do you power WiFi module with good enough current source at 3.3V?

Stoatwblr, 2024-08-01:
Same problem. Happening with 1.3.3 on Windows or Linux
FTDI serial port, 3.3V from a battery, TX/RX lines disconnected from the underlaying TuyaMCU device (PV28 environment sensor)

Tapping CEN doesn't work for the detection procedure to start, only power cycling the CBU

Looking at the serial port, what I see is constant attempts to handshake with the TuyaMCU (disconnected as above)

when CEN is grounded or tapped, output stops 

When released, the following is seen at 115200bps

'<break>
<break>
<break>
<break>
<break x 013>
<break>
<break x 013>
<break>
<break x 013>
<break>
<break x 013>
U<0xaa><break x 004>
<0xff><break>
<break x 013>
U<0xaa><break x 004>
<0xff><break>
<break x 013>
U<0xaa><break x 004>
<0xff><break>
<break x 013>
U<0xaa><break x 004>'

(0xaa, 0xff is the handshaking)

The same sequence is seen if power cycling the CBU

As the CBU is disconnected from the device and the only connections are tx/rx/3.3v/gnd, I suspect this is some form of firmware issue. I now have 3 like this (TH08-AHT20 and another PV28)

NB: The same serial port HAS been used to flash multiple other CBU and ESP devices. The standard problem of its 3.3V being very low current applies, which is why I'm using a separate 3.3V supply or the device's USB input

Failure to set baud rate occurs at 912k, 1.5M and 115k2