openshwprojects/BK7231GUIFlashTool #16: xbuild on Ubuntu fails

staspika, 2023-06-18:
Attempting to build using `xbuild` on Ubuntu results in the following error:

OBKDeviceAPI.cs(125,61): error CS0012: The type `System.Configuration.ConfigurationSection' is defined in an assembly that is not referenced. Consider adding a reference to assembly `System.Configuration, Version=2.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'

Any ideas how to make it build successfully?

LazyGatto, 2023-06-30:
Try `msbuild` instead.

```
sudo apt install gnupg ca-certificates
sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 3FA7E0328081BFF6A14DA29AA6A19B38D3D831EF
echo "deb https://download.mono-project.com/repo/ubuntu stable-bionic main" | sudo tee /etc/apt/sources.list.d/mono-official-stable.list
sudo apt update
sudo apt install mono-roslyn mono-complete mono-dbg msbuild
```

jacobefaust, 2023-07-24:
@LazyGatto !  Your solution worked for me!  Thank you for posting this!