openshwprojects/BK7231GUIFlashTool #18: "The beginning of buffer in UART contains  data."

alexonpeace, 2023-07-30:
Flasher mode: BK7231N
Going to open port: COM3.
Serial port open!
Getting bus... (now, please do reboot by CEN or by power off/on)
Getting bus failed, will try again - 0/100!
Getting bus failed, will try again - 1/100!
Getting bus failed, will try again - 2/100!
Getting bus success!
Going to set baud rate setting (115200)!
failed with serial.BytesToRead 0 (expected 12)
The beginning of buffer in UART contains  data.
Failed to set baud rate!
There was no result to save.

evadim, 2023-10-04:
Looks like I have similar issue with BK7231T.
Read firmware fixed by increasing "cfg_readReplyStyle" from 0 to 5 on "UART Timeots" tab.
Writing fail after erasing.

vitezfh, 2024-02-08:
Thanks for documenting it here! Helped me a lot.