openshwprojects/BK7231GUIFlashTool #19: Unable to backup the BK7231N

S-trace, 2023-08-05:
Hi.

I'm trying to backup the firmware in my BK7231N device (the Wi-Fi switch with CB2S module).

I have soldered 5 wires - GND, VCC, RX, TX, CEN. 
I have provided 3V3 VCC and 3V3 UART I/O voltage.
I have connected the wires as the following:
```
[PC]    [Middleware]    [Switch]
GND     GND             GND
VCC     +5V0=>+3V3 DCDC VCC
RX      RX              TX
TX      TX              RX
NC      NC              CEN
```

But every time I start `BK7231Flasher.exe` it fails with the following error message:
```
Going to start reading at offset 0x00...
Reading 0x00... failed with serial.BytesToRead 4095 (expected 4111)
The beginning of buffer in UART contains 040EFF01E0FCF4061009000000200069 data.
Failed! There was no result to save.
```

How can I fix this?

I tried to "Set baud rate" to 115200, but this does not help - still the same error.

Under Windows 10 I got a bit weirder message - something like "The port does not exist" with the same hardware, but the Windows can see my 3V3 UART converter and assigned its name as com3.

Thanks.

openshwprojects, 2023-08-13:
Hello, sorry for the late reply.
On non-Windows platforms, we've had reports that UART is behaving a bit different but at the moment we are not able to determine what causes it.
On Windows, all ports should be selectable as long as they are not in use. Do you have any other application that may be taking the port in use, maybe Cura or smth?

wasteoinc, 2023-11-10:
I have exactly the same issue with mono in linux with a CB2S board, but the flashing worked as a charm in the command line. So it is some issue with the app. If you need any details will be happy to provide. Thanks for the great project nontheless

vinibali, 2023-12-08:
I had the same problem on Arch Linux. Write was working fine, but it stopped at the backup.
```
failed with serial.BytesToRead 4095 (expected 4111)
```
It might worth a try to use this:
https://github.com/OpenBekenIOT/hid_download_py
Best regards

ntqbit, 2024-03-04:
Same problem here. Reading (neither backup nor OBK read) doesn't work, while writing works on arch linux.
Output:
```
...
SetProtectState(True) success!
Going to start reading at offset 0x00...
Reading 0x00... failed with serial.BytesToRead 4095 (expected 4111)
The beginning of buffer in UART contains 040EFF01E0FCF4061009000000200069 data.
Failed! There was no result to save.
```

[hid_download_py](https://github.com/OpenBekenIOT/hid_download_py) tool works (both reading and writing).