openshwprojects/BK7231GUIFlashTool #21: Writing sector failed

Tony2k, 2023-11-07:
I'm trying to flash a bk7231N smart switch but I have write fails. Changing baud rate only changes the point at which the error occurs, this is the log: 

`All selected sectors erased!
Writing sector 0x00... ok! Writing sector 0x1000... ok! Writing sector 0x2000... ok! Writing sector 0x3000... ok! Writing sector 0x4000... ok! Writing sector 0x5000... ok! Writing sector 0x6000... ok! Writing sector 0x7000... ok! Writing sector 0x8000... ok! Writing sector 0x9000... ok! Writing sector 0xA000... ok! Writing sector 0xB000... ok! Writing sector 0xC000... ok! Writing sector 0xD000... ok! Writing sector 0xE000... ok! Writing sector 0xF000... ok! failed with serial.BytesToRead 1 (expected 15)
The beginning of buffer in UART contains FF data.
Writing sector 0x10000... Writing sector 65536 failed!
Writing file data to chip failed.`

Tried also the change usb port but nothing changed, do you have any idea?

edwardpoll, 2023-11-08:
I am getting the exact same issue

All selected sectors erased!
Going to do erase, start 1904640, sec count 1!
Erasing sector 1904640... ok! 
All selected sectors erased!
Writing sector 0x00... ok! Writing sector 0x1000... ok! Writing sector 0x2000... ok! Writing sector 0x3000... ok! Writing sector 0x4000... ok! Writing sector 0x5000... ok! Writing sector 0x6000... ok! Writing sector 0x7000... ok! Writing sector 0x8000... ok! Writing sector 0x9000... ok! Writing sector 0xA000... ok! Writing sector 0xB000... ok! Writing sector 0xC000... ok! Writing sector 0xD000... ok! Writing sector 0xE000... ok! failed with serial.BytesToRead 14 (expected 15)
The beginning of buffer in UART contains D0FF00000000000000000000000E data.
Writing sector 0xF000... Writing sector 61440 failed!
Writing file data to chip failed.

Flashing a BK7231N CB3S

Tried multiple programmers and usb ports.

samuelolteanu, 2023-11-08:
same with aubess smart swich (2 of them). both itchiptool and openbeken fails to flash.

All selected sectors erased!
Writing sector 0x00... ok! CheckRespond_FlashWrite4K: bad value returned?
Writing sector 0x1000... Writing sector 4096 failed!
Writing file data to chip failed.

Tony2k, 2023-11-08:
Same for me, I'm not able to revive my Aubess smart switch beceuse I can't flash the dump.

samuelolteanu, 2023-11-09:
Failed to write previously flashed tuya smoke detector. Maybe some online dependency is broken?

mariupol1963, 2024-06-14:
The same for me wit smart plug

io2345, 2026-04-26:
For BK7238 (T1-3S):
```
Preparing to write data file to chip - resetting bus and baud...
Getting bus... (now, please do reboot by CEN or by power off/on)
Getting bus success!
Going to set baud rate setting (115200)!
Will try to read device flash MID (for unprotect N):
Flash MID loaded: 152085
Will now search for Flash def in out database...
Flash def found! For: 152085
Flash information: mid: 152085, icName: P25Q16HBK, manufacturer: Puya, szMem: 200000, szSR: 2, cwUnp: 0, cwEnp: 7, cwMsk: 407C, sb: 2, lb: 5, cwdRd: 05-35-FF-FF, cwdWr: 01-31-FF-FF
Flash size is 2MB
Entering SetProtectState(True)...
sr: 0
sr: 0
final sr: 0
msk: 407c
cw: 0, sb: 2, lb: 5
bfd: 0
SetProtectState(True) success!
Going to do erase, start 0x00, sec count 306!
Erasing block 0x00... ok! Erasing block 0x10000... ok! Erasing block 0x20000... ok! Erasing block 0x30000... ok! Erasing block 0x40000... ok! Erasing block 0x50000... ok! Erasing block 0x60000... ok! Erasing block 0x70000... ok! Erasing block 0x80000... ok! Erasing block 0x90000... ok! Erasing block 0xA0000... ok! Erasing block 0xB0000... ok! Erasing block 0xC0000... ok! Erasing block 0xD0000... ok! Erasing block 0xE0000... ok! Erasing block 0xF0000... ok! Erasing block 0x100000... ok! Erasing block 0x110000... ok! Erasing block 0x120000... ok! Erasing sector 0x130000... ok! Erasing sector 0x131000... ok! 
All selected sectors erased!
failed with serial.BytesToRead 0 (expected 15)
The beginning of buffer in UART contains  data.
0x00... Writing sector 0x00 failed!
Writing file data to chip failed.
```