openshwprojects/BK7231GUIFlashTool #22: Can't compile for Linux

thalesmaoa, 2023-12-04:
Hi there, I'm trying to use it in Linux, but I'm having trouble to compile it. Is it compatible with Linux?

openshwprojects, 2023-12-07:
What are the errors? I had reports that users have compiled it with Mono on Linux

deepcoder, 2024-01-25:
Below are the errors I am getting when trying to compile on Debian 12:

```
user@deepmbp:~/OpenBK7231T_App$ xbuild

>>>> xbuild tool is deprecated and will be removed in future updates, use msbuild instead <<<<

XBuild Engine Version 14.0
Mono, Version 6.12.0.200
Copyright (C) 2005-2013 Various Mono authors

Build started 1/24/2024 5:59:00 PM.
__________________________________________________
/home/user/OpenBK7231T_App/openBeken_win32_mvsc2017.vcxproj: error : /home/user/OpenBK7231T_App/openBeken_win32_mvsc2017.vcxproj: /home/user/OpenBK7231T_App/openBeken_win32_mvsc2017.vcxproj could not import "$(VCTargetsPath)\Microsoft.Cpp.Default.props"
Build FAILED.
Errors:

/home/user/OpenBK7231T_App/openBeken_win32_mvsc2017.vcxproj: error : /home/user/OpenBK7231T_App/openBeken_win32_mvsc2017.vcxproj: /home/user/OpenBK7231T_App/openBeken_win32_mvsc2017.vcxproj could not import "$(VCTargetsPath)\Microsoft.Cpp.Default.props"

	 0 Warning(s)
	 1 Error(s)

Time Elapsed 00:00:00.0662420
```

corpix, 2024-02-05:
That's strange, I've just succeeded to build it on NixOS. Full log just in case it will help you to debug the problem:

<details>

<summary>Expand</summary>

```console
 ~/projects/src/github.com/openshwprojects/BK7231GUIFlashTool  λ  xbuild

>>>> xbuild tool is deprecated and will be removed in future updates, use msbuild instead <<<<

XBuild Engine Version 14.0
Mono, Version 6.12.0.182
Copyright (C) 2005-2013 Various Mono authors

Build started 2/5/2024 11:04:12 PM.
__________________________________________________
/home/user/projects/src/github.com/openshwprojects/BK7231GUIFlashTool/BK7231Flasher.sln:  warning : Don't know how to handle GlobalSection ExtensibilityGlobals, Ignoring.
Project "/home/user/projects/src/github.com/openshwprojects/BK7231GUIFlashTool/BK7231Flasher.sln" (default target(s)):
	Target ValidateSolutionConfiguration:
		Building solution configuration "Debug|Any CPU".
	Target Build:
		Project "/home/user/projects/src/github.com/openshwprojects/BK7231GUIFlashTool/BK7231Flasher/BK7231Flasher.csproj" (default target(s)):
			Target PrepareForBuild:
				Configuration: Debug Platform: AnyCPU
				Created directory "bin/Debug/"
				Created directory "obj/Debug/"
			Target CopyFilesMarkedCopyLocal:
				Copying file from '/home/user/projects/src/github.com/openshwprojects/BK7231GUIFlashTool/packages/Newtonsoft.Json.13.0.3/lib/net35/Newtonsoft.Json.dll' to '/home/user/projects/src/github.com/openshwprojects/BK7231GUIFlashTool/BK7231Flasher/bin/Debug/Newtonsoft.Json.dll'
			Target GenerateResources:
				Tool /nix/store/3y3gbf12dg4wz4lqn4qjjnr4bz79cm74-mono-6.12.0.182/lib/mono/4.5/resgen.exe execution started with arguments: /useSourcePath /compile "FormDownloader.resx,obj/Debug/BK7231Flasher.FormDownloader.resources" 
				Tool /nix/store/3y3gbf12dg4wz4lqn4qjjnr4bz79cm74-mono-6.12.0.182/lib/mono/4.5/resgen.exe execution started with arguments: /useSourcePath /compile "FormExtractedConfig.resx,obj/Debug/BK7231Flasher.FormExtractedConfig.resources" 
				Tool /nix/store/3y3gbf12dg4wz4lqn4qjjnr4bz79cm74-mono-6.12.0.182/lib/mono/4.5/resgen.exe execution started with arguments: /useSourcePath /compile "FormMain.resx,obj/Debug/BK7231Flasher.FormMain.resources" 
				Tool /nix/store/3y3gbf12dg4wz4lqn4qjjnr4bz79cm74-mono-6.12.0.182/lib/mono/4.5/resgen.exe execution started with arguments: /useSourcePath /compile "FormOBKConfig.resx,obj/Debug/BK7231Flasher.FormOBKConfig.resources" 
				Tool /nix/store/3y3gbf12dg4wz4lqn4qjjnr4bz79cm74-mono-6.12.0.182/lib/mono/4.5/resgen.exe execution started with arguments: /useSourcePath /compile "FormPin.resx,obj/Debug/BK7231Flasher.FormPin.resources" 
				Tool /nix/store/3y3gbf12dg4wz4lqn4qjjnr4bz79cm74-mono-6.12.0.182/lib/mono/4.5/resgen.exe execution started with arguments: /useSourcePath /compile "FormPrompt.resx,obj/Debug/BK7231Flasher.FormPrompt.resources" 
				Tool /nix/store/3y3gbf12dg4wz4lqn4qjjnr4bz79cm74-mono-6.12.0.182/lib/mono/4.5/resgen.exe execution started with arguments: /useSourcePath /compile "Properties/Resources.resx,obj/Debug/BK7231Flasher.Properties.Resources.resources" 
			Target GenerateSatelliteAssemblies:
			No input files were specified for target GenerateSatelliteAssemblies, skipping.
			Target CoreCompile:
				Tool /nix/store/3y3gbf12dg4wz4lqn4qjjnr4bz79cm74-mono-6.12.0.182/lib/mono/4.5/csc.exe execution started with arguments: /noconfig /debug:full /debug:portable /optimize- /out:obj/Debug/BK7231Flasher.exe /resource:obj/Debug/BK7231Flasher.FormDownloader.resources /resource:obj/Debug/BK7231Flasher.FormExtractedConfig.resources /resource:obj/Debug/BK7231Flasher.FormMain.resources /resource:obj/Debug/BK7231Flasher.FormOBKConfig.resources /resource:obj/Debug/BK7231Flasher.FormPin.resources /resource:obj/Debug/BK7231Flasher.FormPrompt.resources /resource:obj/Debug/BK7231Flasher.Properties.Resources.resources BitUtils.cs BK7231Flasher.cs BKFlashList.cs Colors.cs ConfigBase.cs CRC.cs FormDownloader.cs FormDownloader.Designer.cs FormExtractedConfig.cs FormExtractedConfig.Designer.cs FormMain.cs FormMain.Designer.cs FormMain_Net_Scanner.cs FormMain_Net_SingleDevice.cs FormMain_TuyaConfigExtractor.cs FormOBKConfig.cs FormOBKConfig.Designer.cs FormPin.cs FormPin.Designer.cs FormPrompt.cs FormPrompt.Designer.cs ILogListener.cs MiscUtils.cs ModifyProgressBarColor.cs MySettings.cs OBKConfig.cs OBKDeviceAPI.cs OBKFlags.cs OBKFlashLayout.cs OBKMassBackup.cs OBKRoles.cs OBKScanner.cs Program.cs Properties/AssemblyInfo.cs Rand.cs RequiredLibrariesCheck.cs TuyaConfig.cs RFPartitionUtil.cs RichTextUtil.cs SecurityProtocolTypeExtensions.cs TuyaModules.cs Properties/Resources.Designer.cs Properties/Settings.Designer.cs /target:winexe /win32icon:bk_icon.ico /define:"DEBUG;TRACE" /nostdlib /platform:AnyCPU /reference:../packages/Newtonsoft.Json.13.0.3/lib/net35/Newtonsoft.Json.dll /reference:/nix/store/3y3gbf12dg4wz4lqn4qjjnr4bz79cm74-mono-6.12.0.182/lib/mono/2.0-api/System.dll /reference:/nix/store/3y3gbf12dg4wz4lqn4qjjnr4bz79cm74-mono-6.12.0.182/lib/mono/2.0-api/System.Data.dll /reference:/nix/store/3y3gbf12dg4wz4lqn4qjjnr4bz79cm74-mono-6.12.0.182/lib/mono/gac/System.Deployment/4.0.0.0__b03f5f7f11d50a3a/System.Deployment.dll /reference:/nix/store/3y3gbf12dg4wz4lqn4qjjnr4bz79cm74-mono-6.12.0.182/lib/mono/2.0-api/System.Drawing.dll /reference:/nix/store/3y3gbf12dg4wz4lqn4qjjnr4bz79cm74-mono-6.12.0.182/lib/mono/2.0-api/System.Net.dll /reference:/nix/store/3y3gbf12dg4wz4lqn4qjjnr4bz79cm74-mono-6.12.0.182/lib/mono/2.0-api/System.Windows.Forms.dll /reference: [... cut ...]
 ~/projects/src/github.com/openshwprojects/BK7231GUIFlashTool  λ  xbuild --version                                                                                                                                                                                                                                             

>>>> xbuild tool is deprecated and will be removed in future updates, use msbuild instead <<<<

XBuild Engine Version 14.0
Mono, Version 6.12.0.182
Copyright (C) 2005-2013 Various Mono authors
MSBUILD: error MSBUILD0000: Project file '--version' not found.
```

</section>

fairderben, 2024-12-25:
I suggest closing this issue, as recent builds (including latest [[339f0dc](https://github.com/openshwprojects/BK7231GUIFlashTool/commit/339f0dcd8c23af7f0931bce264132e661a58ddb0)]) build just fine with mono. Thanks for that btw!

tddpirate, 2025-02-16:
I tried to compile commit 277c454 (tagged as 'v8') under Linux (Debian bookworm).
It built a `BK7231Flasher/bin/Debug/Newtonsoft.Json.dll` file (note also the change from documentation - `bin/Debug` rather than `bin/debug`). However, it did not build any `*.exe` file.

The only changes among commits 339f0dc and 277c454 are in *.cs files, and the error messages hint at stuff not found where it was expected to be found.
The problem could be missing configurations or missing Debian mono packages, fixable by improving the documentation rather than the code itself.

The output of my `xbuild` command invocation is in the following attachment.
[xbuild-2025feb16.log](https://github.com/user-attachments/files/18813928/xbuild-2025feb16.log)

List of mono packages installed in my PC (created by `dpkg -l | grep mono`) is as follows.
ii  ca-certificates-mono                           6.8.0.105+dfsg-3.3                           all          Common CA certificates (Mono keystore)
ii  fonts-inconsolata                              001.010-6                                    all          monospace font for pretty code listings and for the terminal
ii  fonts-noto-mono                                20201225-1                                   all          "No Tofu" monospaced font family with large Unicode coverage
ii  fonts-tlwg-mono                                1:0.7.3-1                                    all          Thai TlwgMono font (dependency package)
ii  fonts-tlwg-mono-otf                            1:0.7.3-1                                    all          Thai TlwgMono OpenType font
ii  libmono-btls-interface4.0-cil                  6.8.0.105+dfsg-3.3                           amd64        Mono Mono.Btls.Interface library (for CLI 4.0)
ii  libmono-cairo4.0-cil                           6.8.0.105+dfsg-3.3                           all          Mono Cairo library (for CLI 4.0)
ii  libmono-corlib4.5-cil                          6.8.0.105+dfsg-3.3                           all          Mono core library (for CLI 4.5)
ii  libmono-corlib4.5-dll                          6.8.0.105+dfsg-3.3                           all          Mono core library (for CLI 4.5)
ii  libmono-i18n-cjk4.0-cil                        6.8.0.105+dfsg-3.3                           all          Mono I18N.CJK library (for CLI 4.0)
ii  libmono-i18n-mideast4.0-cil                    6.8.0.105+dfsg-3.3                           all          Mono I18N.MidEast library (for CLI 4.0)
ii  libmono-i18n-other4.0-cil                      6.8.0.105+dfsg-3.3                           all          Mono I18N.Other library (for CLI 4.0)
ii  libmono-i18n-rare4.0-cil                       6.8.0.105+dfsg-3.3                           all          Mono I18N.Rare library (for CLI 4.0)
ii  libmono-i18n-west4.0-cil                       6.8.0.105+dfsg-3.3                           all          Mono I18N.West library (for CLI 4.0)
ii  libmono-i18n4.0-all                            6.8.0.105+dfsg-3.3                           all          Mono I18N libraries metapackage (for CLI 4.0)
ii  libmono-i18n4.0-cil                            6.8.0.105+dfsg-3.3                           all          Mono I18N base library (for CLI 4.0)
ii  libmono-microsoft-build-engine4.0-cil          6.8.0.105+dfsg-3.3                           all          Mono Microsoft.Build.Engine library (for CLI 4.0)
ii  libmono-microsoft-build-framework4.0-cil       6.8.0.105+dfsg-3.3                           all          Mono Microsoft.Build.Framework library (for CLI 4.0)
ii  libmono-microsoft-build-tasks-v4.0-4.0-cil     6.8.0.105+dfsg-3.3                           all          Mono Microsoft.Build.Tasks.v4.0 library (for CLI 4.0)
ii  libmono-microsoft-build-utilities-v4.0-4.0-cil 6.8.0.105+dfsg-3.3                           all          Mono Microsoft.Build.Utilities.v4.0 library (for CLI 4.0)
ii  libmono-security4.0-cil                        6.8.0.105+dfsg-3.3                           all          Mono Security library (for CLI 4.0)
ii  libmono-system-configuration4.0-cil            6.8.0.105+dfsg-3.3                           all          Mono System.Configuration library (for CLI 4.0)
ii  libmono-system-core4.0-cil                     6.8.0.105+dfsg-3.3                           all          Mono System.Core library (for CLI 4.0)
ii  libmono-system-data4.0-cil                     6.8.0.105+dfsg-3.3                           all          Mono System.Data library (for CLI 4.0)
ii  libmono-system-drawing4.0-cil                  6.8.0.105+dfsg-3.3                           all          Mono System.Drawing library (for CLI 4.0)
ii  libmono-system-enterpriseservices4.0-cil       6.8.0.105+dfsg-3.3                           all          Mono System.EnterpriseServices library (for CLI 4.0)
ii  libmono-system-numerics4.0-cil                 6.8.0.105+dfsg-3.3                           all          Mono System.Numerics library (for CLI 4.0)
ii  libmono-system-runtime-serialization4.0-cil    6.8.0.105+dfsg-3.3                           all          Mono System.Runtime.Serialization Library (for CLI 4.0)
ii  libmono-system-security4.0-cil                 6.8.0.105+dfsg-3.3                           all          Mono System.Security library (for CLI 4.0)
ii  libmono-system-servicemodel-internals0.0-cil   6.8.0.105+dfsg-3.3                           all          Mono System.ServiceModel.Internals Library (for CLI 4.0)
[... cut ...]
ii  mono-runtime                                   6.8.0.105+dfsg-3.3                           amd64        Mono runtime - default version
ii  mono-runtime-common                            6.8.0.105+dfsg-3.3                           amd64        Mono runtime - common files
ii  mono-runtime-sgen                              6.8.0.105+dfsg-3.3                           amd64        Mono runtime - SGen
ii  mono-xbuild                                    6.8.0.105+dfsg-3.3                           all          MSBuild-compatible build system for Mono

tddpirate, 2025-12-01:
Balucio,
- Which Linux distribution did you use?
- From where did you get the Mono libraries and whatever you needed to build the software?