openshwprojects/BK7231GUIFlashTool #23: WB2-12F development boards

MacSass, 2023-12-31:
Hello,
I ordered two WB2-12F development boards, hoping I would be able to flash them with OpenBeken.
This is the board: https://de.aliexpress.com/item/1005004911487557.html 

Unfortunately I did not yet manage to get them flashed through USB port. 
I see FlashTool opening the port and get the message to "Getting Buss ...(now, please do reboot by CEN or by power off(on)" but even when I connect the board EN pin to ground, or use the on board "burn" button ... nothing else happens.

So my question is: Can those boards be flashed with OpenBeken with the FlashTool and can it be done through the build in USB port?

Thanks - MacSass

openshwprojects, 2023-12-31:
You have BL602, and this tool is for BK currently.
You should rather follow: https://www.elektroda.com/rtvforum/topic3889041.html

MacSass, 2023-12-31:
Thanks - I saw that post, but I was hoping it was possible to flash my board through the on-board USB/Serial converter ... but that does not seem to work?
The board mentioned in the post linked above is very different and does not have an on-board USB converter.
I will try connecting my own serial converter to the respective PINs on my board directly to see if I can flash it that way ...