openshwprojects/BK7231GUIFlashTool #32: Bricked WB3S module

3xtant, 2024-05-26:
Hello, as the title says, I managed to brick a WB3S module. I wanted to erase it completely, selected the advanced options, and erased it all, including the RF section. I noticed that successive flashes did not erase options I had made before, which was a problem for me while creating a custom template for a new device with trial-and-error.

After that, the chip never enters flash mode. I cannot read the chip id either. I have used an FTDI USB adapter and a programmer that looks like this https://cdn.hackaday.io/images/200281590474264895.jpg

Are there any suggestions to recover from this? Is my module done for?

P.S. I have backups of the factory firmware and the last working OpenBK.

openshwprojects, 2024-05-26:
https://www.elektroda.com/rtvforum/topic3931424.html

3xtant, 2024-05-26:
Of course, thank you!
I think I have an arduino uno sitting somewhere already configured as an SPI burner. I'll try and report.

3xtant, 2024-05-26:
Marked the issue as closed. Even if the SPI method does not work, I've come to terms with it 😃
Thanks again!

closed 2024-05-26 by 3xtant

3xtant, 2024-06-01:
I just wanted to update here as promised. I was not able to revive my WB3S. I did not use the Arduino as SPI controller, but an rpi pico, where it was super simple to load and modify your python code. I needed to make a few changes to be able to work within the confines of the pico (less than 2MB of storage for one), but the code worked well in very short order.

I read the comments in electroda and several people mentioned issues with the baud rate. I tried many different values and sometimes got semi valid replies, but I was always unable to get a chip id. I have some ideas to automate the baud rate adjustment and go from there, but for this particular project I will simply replace the WB3S with an ESP12F.

It was fun to troubleshoot, but I do want to get my device back in working order.

Thanks again @openshwprojects , this is an awesome project. I am very grateful to you for dedicating so much time and effort to it.