openshwprojects/BK7231GUIFlashTool #33: Failed to open serial port

broyuken, 2024-06-05:
Hello,

I'm fairly sure I'm doing this correctly but I am unable to open the serial connection to flash a CB3S with OBK.
I am using this USB to TTL module
https://www.aliexpress.us/item/3256805987170518.html

and wired up the 4 ports to VCC, GND, and flipped RX and TX. I have also verified that the device is getting power, both with a multimeter, and the fact that it shows up as online in the SmartLife app. 

However I am getting the following error from the flasher whenever I try to do anything.
```
Starting flash new (no backup)!
Now is: Wednesday, June 5, 2024 10:43:24.
Flasher mode: BK7231N
Going to open port: COM4.
Serial port open exception: System.IO.IOException: A device attached to the system is not functioning.

   at System.IO.Ports.InternalResources.WinIOError(Int32 errorCode, String str)
   at System.IO.Ports.SerialStream.InitializeDCB(Int32 baudRate, Parity parity, Int32 dataBits, StopBits stopBits, Boolean discardNull)
   at System.IO.Ports.SerialStream..ctor(String portName, Int32 baudRate, Parity parity, Int32 dataBits, StopBits stopBits, Int32 readTimeout, Int32 writeTimeout, Handshake handshake, Boolean dtrEnable, Boolean rtsEnable, Boolean discardNull, Byte parityReplace)
   at System.IO.Ports.SerialPort.Open()
   at BK7231Flasher.BK7231Flasher.openPort() in E:\GitHub\BK7231GUIFlashTool\BK7231Flasher\BK7231Flasher.cs:line 95
Failed to open serial port!
```

The COM port does show up in Windows so it doesn't appear to be a driver issue. 
[image]

Anyone have an idea what could be the issue here?

joyanes97, 2024-06-07:
Good morning @broyuken , I have encountered the same problem, were you able to solve it?

broyuken, 2024-06-07:
No, I’m hoping someone can chime in and help me troubleshoot.

trurlo, 2025-03-22:
I came across the same issue.  What helped: disconnecting VCC, TX and RX, then, while the app was waiting for the port to open, reconnecting first RX then TX - upon which the app picked up and finished flashing without problems.  VCC was not connected at all until the end.  MCU is CBU on a QS-WIFI-D06-TRIAC-C2

VkV1zion, 2025-09-06:
so ?