openshwprojects/BK7231GUIFlashTool #35: Feature request: Extract power calibration data

dwmw2, 2024-08-20:
(This is a "duplicate" of https://github.com/libretiny-eu/ltchiptool/issues/38 )

I have a bunch of Tongou DIN rail power meters (https://www.elektroda.com/rtvforum/viewtopic.php?p=21191153#21191153),. and have finally got round to liberating them.

I've been experimenting with them, and found that the calibration parameters for the BL0942 chip are available both in the DPs that the Tuya firmware reports, and also in the flash. More details at https://github.com/esphome/issues/issues/6155

So apart from the one I *already* converted without reading its flash first, I can liberate them without losing the factory calibration. It would be nice if the tooling could help new users get this right.

Example firmware dump at http://david.woodhou.se/tongou-tuyafw.bin (sha256 a0b0a2804946f8760f5be0360a3b0325d204087c6aae6c7917297cba2f1341ba)

I found the calibration at 0x001d5000:
```
001d5000  60 3e 00 00 82 30 00 00  13 0c 00 00 5d 0a 00 00  |`>...0......]...|
```

That's
  - UREF = 0x3e60 = 15968
  - IREF = 0x3082 = 12418 (times ten to make 124180)
  - PREF = 0xc13 = 3091 (divide by ten to make 309.1)
  - EREF = 0xa5d = 2653
  
We can apply a sanity check to those, as the PREF can be derived from UREF/IREF, and EREF from PREF as described in the ESPHome ticket above.

dwmw2, 2024-08-27:
Now documented for ESPHome: https://devices.esphome.io/devices/Tongou-TO-Q-SY1-JWT

divadiow, 2025-06-15:
I'd love for this to somehow be used to calib OBK automatically. hmm