openshwprojects/BK7231GUIFlashTool #38: Request: Node-RED tutorial video

daxliniere, 2024-09-02:
Hello,
It would be great if there was a tutorial showing how this can be used for simple switch devices using Node-RED.

Best regards,
Dax.

MacSass, 2024-09-02:
Hi, I don´t know for Node-RED, but as OpenBK in many respects uses the same commands as Tasmota, for mobile phones you can use the "Tasmotrol" app for Android or iOS, they support OpenBK as well.
For Node-RED unfortunately I can´t help - but I think it would be along the lines of other Tasmota devices.

daxliniere, 2024-09-02:
Thanks @MacSass, my main thing was wanting to understand what commands to send to a device like this to turn them on/off:
https://www.elektroda.com/rtvforum/topic4018747.html

[image]

MacSass, 2024-09-02:
I never worked with MQTT, but I guess those should be the same commands like Tasmota:
https://tasmota.github.io/docs/Commands/
Check that link for commands specific for MQTT, it has a section on sending commands via MQTT:
https://tasmota.github.io/docs/MQTT/#configure-mqtt-using-backlog

daxliniere, 2024-09-02:
Thanks so much, @MacSass. I was able to get it working with a Tasmota plugin for Node-RED.

Once the node was set as shown below, all I needed to do was send boolean True/False to turn the device on/off. This is great! I've finally got it working as I needed with your help!🙏

[image]