openshwprojects/BK7231GUIFlashTool #48: BK7231N "Failed to extract keys" on Tuya config template extraction

vinibali, 2025-01-03:
Hi there!
I was following this tutorial: [How to automatically configure GPIO of OpenBeken device? Tuya Config template extraction, BK7231 ](https://www.youtube.com/watch?v=WunlqIMAdgw)
But unfortunately the Flasher was not able to extract the configuration for me, ended up with the following message:
```
Failed to extract keys
```
I'm about to insert the config file, I'm also running on Arch Linux with mono.
[BK7231N_TuyaConfig.zip](https://github.com/user-attachments/files/18303510/BK7231N_TuyaConfig.zip)

divadiow, 2025-01-05:
[t34assignments.txt](https://github.com/user-attachments/files/18309672/t34assignments.txt)
[t34storage.txt](https://github.com/user-attachments/files/18309673/t34storage.txt)

divadiow, 2025-10-21:
the latest Easy Flasher should work on T34s