openshwprojects/BK7231GUIFlashTool #50: Feature request: support for RTL87X0C

java4tech, 2025-01-16:
OpenBeken release 1.18.20 now has the support for RTL97X0C, any plans to add support in the GUI Flasher?

divadiow, 2025-12-26:
Easy Flasher now supports AmebaZ2 read and write

LynxChaus, 2026-03-01:
Unstable support. Reading is working (stable on lower speeds). 
Writing is broken - it's always fail (no matter of what is writing - full flash, or only OBK config partition):

```
Now is: 01 March 2026 22:18:06.
Flasher mode: RTL87X0C
Going to open port: /dev/ttyUSB7.
Port ready!
Starting read...
Read parms: start 0x1F0000 (sector 496), len 0x10000 (16 sectors)
Setting baud rate to 115200
Reading at 0x1F0000... Reading at 0x1F1000... Reading at 0x1F2000... Reading at 0x1F3000... Reading at 0x1F4000... Reading at 0x1F5000... Reading at 0x1F6000... Reading at 0x1F7000... Reading at 0x1F8000... Reading at 0x1F9000... Reading at 0x1FA000... Reading at 0x1FB000... Reading at 0x1FC000... Reading at 0x1FD000... Reading at 0x1FE000... Reading at 0x1FF000... 
Getting hash...
Hash matches 28D6D651C474C551F768A21E18A6BA7CA89DB0163B659237563888A113A11000!
Read complete!
Setting baud rate to 115200
OBK config loaded. You can now view it by clicking 'Change OBK settings' button.
You can also edit it whatever you want.
You can also use 'Write OBK config' button to write it back with your changes.

Starting write!
Write parms: start 0x1F0 (sector 0), len 0x10000 (496 sectors)
Now is: 01 March 2026 22:19:40.
Flasher mode: RTL87X0C
Going to open port: /dev/ttyUSB7.
Port ready!
Setting baud rate to 115200
Now will also write OBK config...
Long name from CFG: OpenRTL87X0C_B7CBCF29
Short name from CFG: rtl87x0CB7CBCF29
Web Root from CFG: https://openbekeniot.github.io/webapp/
Writing at 0x1F0000...Writing OBK config data to chip failed.

```
misleading log message: ```Write parms: start 0x1F0 (sector 0), len 0x10000 (496 sectors)``` - 0x1F0000 truncated to 0x1F0. Also, which ```sector 0``` referenced here?  

File write command sent with wrong VIRTUAL mapped address ```fwd 0 1 981f0000``` instead of ```fwd 0 1 1f0000``` raw flash offset. After that - chip is hung, reset is required. 
FlashTransmit() called with offset 0x981f0000 when writing 0x10000 bytes. Curiously what data at that Xmodem send to..

So, **ltchiptool** still need for writing AmebaZ2 devices.

LynxChaus, 2026-03-06:
I'm can't build on linux mono apps. If you build it for me - i try test it.

divadiow, 2026-03-06:
there's a built exe here with the rtl2 branch code, if that's usable to you. I don't know much about mono :)

https://github.com/divadiow/BK7231GUIFlashTool/actions/runs/22735048808