openshwprojects/BK7231GUIFlashTool #51: Writing sector 0x2B000... Writing sector 176128 failed!

shockinsid, 2025-01-20:
Writing fails on B7231N.

I'm unable to write the backup to the chip or the new firmware. Any advice?

Starting flash new (no backup)!
Now is: 20 January 2025 22:18:22.
Flasher mode: BK7231N
Going to open port: COM6.
Serial port open!
Getting bus... (now, please do reboot by CEN or by power off/on)
Getting bus success!
Going to set baud rate setting (921600)!
Will try to read device flash MID (for unprotect N):
Flash MID loaded: 1560EB
Will now search for Flash def in out database...
Flash def found! For: 1560EB
Flash information: mid: 1560EB, icName: TH25Q16HB, manufacturer: TH, szMem: 1000000, szSR: 2, cwUnp: 0, cwEnp: 7, cwMsk: 407C, sb: 2, lb: 5, cwdRd: 05-35-FF-FF, cwdWr: 01-FF-FF-FF
Entering SetProtectState(True)...
sr: 80
sr: 680
final sr: 680
msk: 407c
cw: 0, sb: 2, lb: 5
bfd: 0
SetProtectState(True) success!
Going to read encryption key...
Encryption key read done!
Encryption key: 510fb093 a3cbeadc 5993a17e c7adeb03
Reading file firmwares/OpenBK7231N_QIO_1.18.21.bin...
Loaded 1220464 bytes from firmwares/OpenBK7231N_QIO_1.18.21.bin...
Using hack to write QIO - just skip bootloader...
... so bootloader will not be overwritten!
Preparing to write data file to chip - resetting bus and baud...
Getting bus... (now, please do reboot by CEN or by power off/on)
Getting bus success!
Going to set baud rate setting (921600)!
Will try to read device flash MID (for unprotect N):
Flash MID loaded: 1560EB
Will now search for Flash def in out database...
Flash def found! For: 1560EB
Flash information: mid: 1560EB, icName: TH25Q16HB, manufacturer: TH, szMem: 1000000, szSR: 2, cwUnp: 0, cwEnp: 7, cwMsk: 407C, sb: 2, lb: 5, cwdRd: 05-35-FF-FF, cwdWr: 01-FF-FF-FF
Entering SetProtectState(True)...
sr: 80
sr: 680
final sr: 680
msk: 407c
cw: 0, sb: 2, lb: 5
bfd: 0
SetProtectState(True) success!
Going to do erase, start 69632, sec count 281!
Erasing sector 69632... ok! Erasing sector 73728... ok! Erasing sector 77824... ok! Erasing sector 81920... ok! Erasing sector 86016... ok! Erasing sector 90112... ok! Erasing sector 94208... ok! Erasing sector 98304... ok! Erasing sector 102400... ok! Erasing sector 106496... ok! Erasing sector 110592... ok! Erasing sector 114688... ok! Erasing sector 118784... ok! Erasing sector 122880... ok! Erasing sector 126976... ok! Erasing sector 131072... ok! Erasing sector 135168... ok! Erasing sector 139264... ok! Erasing sector 143360... ok! Erasing sector 147456... ok! Erasing sector 151552... ok! Erasing sector 155648... ok! Erasing sector 159744... ok! Erasing sector 163840... ok! Erasing sector 167936... ok! Erasing sector 172032... ok! Erasing sector 176128... ok! Erasing sector 180224... ok! Erasing sector 184320... ok! Erasing sector 188416... ok! Erasing sector 192512... ok! Erasing sector 196608... ok! Erasing sector 200704... ok! Erasing sector 204800... ok! Erasing sector 208896... ok! Erasing sector 212992... ok! Erasing sector 217088... ok! Erasing sector 221184... ok! Erasing sector 225280... ok! Erasing sector 229376... ok! Erasing sector 233472... ok! Erasing sector 237568... ok! Erasing sector 241664... ok! Erasing sector 245760... ok! Erasing sector 249856... ok! Erasing sector 253952... ok! Erasing sector 258048... ok! Erasing sector 262144... ok! Erasing sector 266240... ok! Erasing sector 270336... ok! Erasing sector 274432... ok! Erasing sector 278528... ok! Erasing sector 282624... ok! Erasing sector 286720... ok! Erasing sector 290816... ok! Erasing sector 294912... ok! Erasing sector 299008... ok! Erasing sector 303104... ok! Erasing sector 307200... ok! Erasing sector 311296... ok! Erasing sector 315392... ok! Erasing sector 319488... ok! Erasing sector 323584... ok! Erasing sector 327680... ok! Erasing sector 331776... ok! Erasing sector 335872... ok! Erasing sector 339968... ok! Erasing sector 344064... ok! Erasing sector 348160... ok! Erasing sector 352256... ok! Erasing sector 356352... ok! Erasing sector 360448... ok! Erasing sector 364544... ok! Erasing sector 368640... ok! Erasing sector 372736... ok! Erasing sector 376832... ok! Erasing sector 380928... ok! Erasing sector 385024... ok! Erasing sector 389120... ok! Erasing sector 393216... ok! Erasing sector 397312... ok! Erasing sector 401408... ok! Erasing sector 405504... ok! Erasing sector 409600... ok! Erasing sector 413696... ok! Erasing sector 417792... ok! Erasing sector 421888... ok! Erasing sector 425984... ok! Erasing sector 430080... ok! Erasing sector 434176... ok! Erasing sector 438272... ok! Erasing sector 442368... ok! Erasing sector 446464... ok! Erasing sector 450560... ok! Erasing sector 454656... ok! Erasing sector 458752... ok! Erasing sector 462848... ok! Erasing sector 466944... ok! Erasing sector 471040... ok! Erasing sector 475136... ok! Erasing sector 479232... ok! Erasing sector 483328... ok! Erasing sector 487424... ok! Erasing sector 491520... ok! Erasing sector 495616... ok! Erasing sector 499712... ok! Erasing sector 503808... ok! Erasing sector 507904... ok! Erasing sector 512000... ok! Erasing sector 516096... ok! Erasing sector 520192... ok! Erasing sector 524288... ok! Erasing sector 528384... ok! Erasing sector 532480... ok! Erasing sector 536576... ok! Erasing sector 540672... ok! Erasing sector 544768... ok! Erasing sector 548864... ok! Erasing sector 552960... ok! Erasing sector 557056... ok! Erasing sector 561152... ok! Erasing sector 565248... ok! Erasing sector 569344... ok! Erasing sector 573440... ok! Erasing sector 577536... ok! Erasing sector 581632... ok! Erasing sector 585728... ok! Erasing sector 589824... ok! Erasing sector 593920... ok! Erasing sector 598016... ok! Erasing sector 602112... ok! Erasing sector 606208... ok! Erasing sector 610304... ok! Erasing sector 614400... ok! Erasing sector 618496... ok! Erasing sector 622592... ok! Erasing sector 626688... ok! Erasing sector 630784... ok! Erasing sector 634880... ok! Erasing sector 638976... ok! Erasing sector 643072... ok! Erasing sector 647168... ok! Erasing sector 651264... ok! Erasing sector 655360... ok! Erasing sector 659456... ok! Erasing sector 663552... ok! Erasing sector 667648... ok! Erasing sector 671744... ok! Erasing sector 675840... ok! Erasing sector 679936... ok! Erasing sector 684032... ok! Erasing sector 688128... ok! Erasing sector 692224... ok! Erasing sector 696320... ok! Erasing sector 700416... ok! Erasing sector 704512... ok! Erasing sector 708608... ok! Erasing sector 712704... ok! Erasing sector 716800... ok! Erasing sector 720896... ok! Erasing sector 724992... ok! Erasing sector 729088... ok! Erasing sector 733184... ok! Erasing sector 737280... ok! Erasing sector 741376... ok! Erasing sector 745472... ok! Erasing sector 749568... ok! Erasing sector 753664... ok! Erasing sector 757760... ok! Erasing sector 761856... ok! Erasing sector 765952... ok! Erasing sector 770048... ok! Erasing sector 774144... ok! Erasing sector 778240... ok! Erasing sector 782336... ok! Erasing sector 786432... ok! Erasing sector 790528... ok! Erasing sector 794624... ok! Erasing sector 798720... ok! Erasing sector 802816... ok! Erasing sector 806912... ok! Erasing sector 811008... ok! Erasing sector 815104... ok! Erasing sector 819200... ok! Erasing sector 823296... ok! Erasing sector 827392... ok! Erasing sector 831488... ok! Erasing sector 835584... ok! Erasing sector 839680... ok! Erasing sector 843776... ok! Erasing sector 847872... ok! Erasing sector 851968... ok! Erasing sector 856064... ok! Erasing sector 860160... ok! Erasing sector 864256... ok! Erasing sector 868352... ok! Erasing sector 872448... ok! Erasing sector 876544... ok! Erasing sector 880640... ok! Erasing sector 884736... ok! Erasing sector 888832... ok! Erasing sector 892928... ok! Erasing sector 897024... ok! Erasing sector 901120... ok! Erasing sector 905216... ok! Erasing sector 909312... ok! Erasing sector 913408... ok! Erasing sector 917504... ok! Erasing sector 921600... ok! Erasing sector 925696... ok! Erasing sector 929792... ok! Erasing sector 933888... ok! Erasing sector 937984... ok! Erasing sector 942080... ok! Erasing sector 946176... ok! Erasing sector 950272... ok! Erasing sector 954368... ok! Erasing sector 958464... ok! Erasing sector 962560... ok! Erasing sector 966656... ok! Erasing sector 970752... ok! Erasing sector 974848... ok! Erasing sector 978944... ok! Erasing sector 983040... ok! Erasing sector 987136... ok! Erasing sector 991232... ok! Erasing sector 995328... ok! Erasing sector 999424... ok! Erasing sector 1003520... ok! Erasing [... cut ...] 8000... ok! Writing sector 0x19000... ok! Writing sector 0x1A000... ok! Writing sector 0x1B000... ok! Writing sector 0x1C000... ok! Writing sector 0x1D000... ok! Writing sector 0x1E000... ok! Writing sector 0x1F000... ok! Writing sector 0x20000... ok! Writing sector 0x21000... ok! Writing sector 0x22000... ok! failed with serial.BytesToRead 2 (expected 15)
The beginning of buffer in UART contains 0000 data.
Writing sector 0x23000... Writing sector 143360 failed!
Writing file data to chip failed.

hllhll, 2025-06-01:
I have similar issues, looks like the uart interface only works for 8-10 seconds on a battery powered CB3S temperature sensor ; it fails to read on some block. mid reading. same go if I only try to flash(write). It erases, starts to write then fails somewhere in the middle
I will try to partially reprogram blocks individually through another tool or something

hllhll, 2025-06-01:
> Battery powered sensors may use TuyaMCU where the MCU cuts off power after some time. That is the reason it may fail. Using different program won't help. Try powering CB3S directly.

I read a couple of threads and I already did that and it still hang after a few seconds.
HOWEVER, I hooked up VCC directly and left ground on the battery terminal.
After reading ur post again I consider - what the hack, let's also wire GND directly; now it FLASHES.
WHY ON EARTH Someone disconnect gnd :/

BTW Another question, can this low-power be controller by code somehow? does the "mystery chip" does that?